Duet3D/RepRapFirmware #18: Not sending data to aux serial port.

mirage335, 2015-11-24:
Messages sent by the PanelDue on the aux serial port show up in debugging output on the USB serial port. However, responses generally do not seem to be sent to the same. Additionally, messages sent on the PanelDue console "M114" do not show up on the same. Only the "Connecting.../Idle" and uptime information seems to reach the PanelDue.

Oscilloscope probing seems to confirm this, showing messages on the blue, but not green, wire. SImilarly, disconnecting the green wire still allows the PanelDue to send messages, but not receive any status updates, including "Connecting.../Idle" or uptime.

Is there some g-code needed for configuring this with the new firmware? For diagnostic use, being able to send abritrary messages to arbitrary serial ports would also be useful.

mirage335, 2015-11-24:
Sending the "M20 S2" command manually from the console actually gets a reply on the AUX port, and subsequently, files show up in the SD card menu. Nothing shows up in the console though, nor by default.

dc42, 2015-11-25:
That is expected behaviour. Any JSON-formatted response will get decoded
and processed in the same way whether the command that provoked it was
sent manually from the console or automatically by the PanelDue.

David Crocker, Escher Technologies Ltd.
http://www.eschertech.com
Tel. +44 (0)20 8144 3265 or +44 (0)7977 211486

dc42, 2015-11-25:
The response to M114 and similar commands is stored in the response
buffer in the Duet and gets picked up by the PanelDue next time it sends
the M408 poll command. So there will be a short delay between sending
M114 and seeing the response in the console.

If you send a command such as M114 or G32 from the web interface, the
response will also be picked up by PanelDue.

The only gcode you need to add to config.g on the Duet us the M575
command to set the baud rate an requires a valid checksum. The suggested
command is:

M575 P1 B57600 S1                    ; Comms parameters for PanelDue

David Crocker, Escher Technologies Ltd.
http://www.eschertech.com
Tel. +44 (0)20 8144 3265 or +44 (0)7977 211486

mirage335, 2015-11-25:
Actually, I found the problem. Apparently, valid responses are not sent to the PanelDue unless the tools (ie. extruders) are defined with M563/G10 commands.

Perhaps that problem doesn't apply to most users, since I am still setting up an unusually complex machine (ie. TazMega) with non-extruder (ie. laser, spindle) tools.

closed 2015-12-18 by mirage335