Duet3D/RepRapFirmware #19: Limited functionality with absolute extrusion distances

tomasf, 2015-12-11:
When printing G-code with non-relative extrusion distances, some features like filament usage and estimated time left based on filament usage, stop working properly. Filament usage works for a while, but resets periodically, presumably because the G-code uses `G92 E0` before every layer change. This also causes the web interface progress bar to stop working, because it uses the used filament length. `timesLeft.filament` in `/rr_status` is always zero.

These things should either be fixed (or absolute extrusion be declared unsupported).

Tested using Slic3r 1.2.9 with a Kossel Mini (T3P3 R3).

Here's the same print with both absolute and relative extrusion distances: [example_absolute+relative.zip](https://github.com/dc42/RepRapFirmware/files/59395/example_absolute.relative.zip)

dc42, 2015-12-11:
This is a known issue that I intend to fix in the next release.

dc42, 2015-12-27:
Fixed in version 1.09o.

closed 2015-12-27 by dc42