Duet3D/RepRapFirmware #22: File uploads through web interface sometimes hang at 100%

crispy1066, 2015-12-15:
Possible theory is it's expecting more data than really exists.  My guess is webserver.cpp lines 500-660.

shadow651, 2015-12-15:
I can also confirm that I occasionally experience this issue.
I typically cancel the upload and restart; it usually works the second time.

criticallimit, 2015-12-18:
I can confirm this too.
It happens to me nearly every upload bigger than 3GB. Sometimes it works at the second try, but most of the time it stops completely at no special percentage. After that there are no files shown on the card anymore in the webserver. Only reset helps.

dc42, 2015-12-27:
Please try version 1.09o.

criticallimit, 2015-12-28:
new version made it worse.
No files are uploaded now to the end. All files that I´m uploading stuck at no specific percentage.
Only very small files are going through. But files bigger than 10MB are stucking. Most of my files are 20-100MB.

dc42, 2015-12-28:
Thanks for your feedback. I have had one other report of a similar problem and I will do further testing over the next few days. Meanwhile I suggest you either try a different SD card, or revert to 1.09m.

criticallimit, 2015-12-28:
Have tested it with 5 different SD cards now which are working fine in all other hardware (phone, smoothie, camera etc). None of them working in the duet. But I think it´s not a SD card problem. It´s the webserver.

W3DRK, 2016-01-18:
Just had this happen yesterday with 1.09r on a Duet v0.6.  After two failed attempts, I re-homed all axis and it worked fine.  Coincidence?

tomasf, 2016-02-15:
I've often seen this one, too. I now upload everything over FTP, which works much better.

dc42, 2016-02-15:
It would be helpful if you can supply a Wireshark trace of a failed upload.

David Crocker, Escher Technologies Ltd.
http://www.eschertech.com
Tel. +44 (0)20 8144 3265 or +44 (0)7977 211486

WZ9V, 2016-02-15:
i noticed similar problems last night when I was manually calibrating using the web tool over a three hour period.  At various times I would get errors which are hard to reproduce.

AJAX timeout (reconnect usually fixed it)
File upload lockout (retry after disconnect/reconnect usually fixed it)
Web interface looking connected but no g-code works (disconnect/reconnect usually fixed it)

I was using Firefox on Linux Mint and it had some recent updates that make me wonder if it's browser issues. I'll try Chrome today and see if it's better.

dc42, 2016-12-20:
There have been numerous improvements to the network code since this report was filed and I have had no reports of this issue in recent months. So I am closing it. Feel free to re-open it if you experience it using firmware 1.16 with DWC 1.13, or later versions.

closed 2016-12-20 by dc42