Duet3D/RepRapFirmware #23: Roland.h and related files not checked in?

dcnewman, 2015-12-28:
RepRapFirmware.h has references to Roland.h and a type defined by it.  However, Roland.h appears nowhere in the repository.

dcnewman, 2015-12-28:
Looks like Pins.h wasn't checked in either.

dc42, 2015-12-28:
Sorry about that. I've now checked in Pins.h, Roland.h and Roland.cpp.

David Crocker, Escher Technologies Ltd.
http://www.eschertech.com
Tel. +44 (0)20 8144 3265 or +44 (0)7977 211486

dcnewman, 2015-12-28:
Looks like they haven't been sync'd up to the publicly visible github repo?  At least, I do not yet see those three files in the dev branch.  (Nor the master branch --  I know it's obsolete, but I checked there just in case.)

dc42, 2015-12-28:
Sorry, I must have forgotten to sync it. Will do so when I am back in the office. You can get Roland.h and .cpp from chrishamm's fork.

closed 2016-01-17 by dcnewman