Duet3D/RepRapFirmware #25: Panel due is locked with 1.09o

criticallimit, 2015-12-28:
My PanelDue is locked in the the new Version. Tapping on the screen is doing nothing. All Values are ok and are updatet, but if you klick on a button nothing happens. Like it´s locked.
Did a reset of the Panel and it was going to the print screen, but you cant click. Every time I do a reset it´s going back to the print screen updating all values correctly but the buttons are locked.

thevisad, 2015-12-28:
I had a similar issue with my PanelDue after an update to 1.09n and flashing the PanelDue to the latest firmware. However, I could not proceed past the screen calibration. Did you also flash your PanelDue firmware as well?

criticallimit, 2015-12-28:
Yes, Panel is up to date (1.13)

thevisad, 2015-12-28:
What screen are you using and how are you powering the PanelDue?

criticallimit, 2015-12-28:
standard screen 4.7" from www.think3dprint3d.com.
Powered from the Duet with standard cable.
Working without any issues before 1.09o

dc42, 2015-12-29:
Please try reverting to 1.09m. There are some issues with code from another developer that I merged in and they could be causing these problems.

BrianGilbert, 2016-01-28:
Assume this can be closed now that your up to r release?

dc42, 2016-01-28:
Believed to be caused by buffer starvation occurring in the new output buffer code in 1.09o. Fixed in 1.09r.

closed 2016-01-28 by dc42