Duet3D/RepRapFirmware #28: Reset upon calculating bed compensation.

mirage335, 2016-01-21:
Crash occurrs when "G30 P3 S0 Z-10000" is executed.Simply reporting the data, without calculating bed compensation, works normally. Executing "G30 P3 S-1 Z-10000" returns valid data - "Bed probe heights: -0.350 -0.387 -0.335 -0.237, mean -0.327, deviation from mean 0.055".

Notably, this did not occurr prior to installing a new tool platform (MightyTool). May be related to the use of positive X/Y coordinate offsets with the differential optical probe - ie. "G31 C0 P500 X40.8 Y8.7 Z0.35", instead of "G31 C0 P500 X0 Y0 Z1.3".

mirage335, 2016-01-21:
As a workaround, three-factor calibration - ie. "G30 P3 S3 Z-10000" seems to work without errors.

dc42, 2016-01-21:
Thanks for your report. Please post your bed.g file, along with the G31
and M558 lines from your config.g file. I have a report that the
firmware may crash if the bed probe points are not ordered so as to be
clockwise around the bed. The issue you report may be related.

David Crocker, Escher Technologies Ltd.
http://www.eschertech.com
Tel. +44 (0)20 8144 3265 or +44 (0)7977 211486

mirage335, 2016-01-22:
Relevant G-Code files are publicized on GitHub.

https://github.com/mirage335/TazMega-SDCard/blob/9a1a282ea1e09dac060fad7264768e8dc59a0238/sys/config.g
https://github.com/mirage335/TazMega-SDCard/blob/9a1a282ea1e09dac060fad7264768e8dc59a0238/sys/bed.g

Please note the bed.g sequence is clockwise, and was successfully used with an earlier set of tools (ie. when the probe was not assigned positive X/Y coordinates).