Duet3D/RepRapFirmware #32: Feature request: Abort homing if bed probe is activated

gavanfantom, 2016-02-16:
I recently updated to 1.09r on my fisher-beta printer and hadn't spotted that the motor directions were reversed between fisher-beta and fisher-1. The first attempt at homing was a disastrous high-speed crash of the hot end and effector into the bed, resulting in a couple of broken (delta, not human) arms. I would dearly love to save other people and their printers from the same fate.

It strikes me that there is no reason that the bed probe should ever be activated during homing and that there's a good chance that the effector would be central enough on homing that the bed probe would be activated if the motors all went in the wrong direction.

There's probably not much that can be done in the scenario where the motors are going in different directions, but that's less likely to occur after a firmware update - it should be caught in the original commissioning where you're more likely to expect this sort of problem and take precautions.

So I'd like to request a feature that if the bed probe is activated at any time during homing, the homing process be immediately aborted and all motion to cease.

It may also be useful for the bed probe to cause the printing to stop during a normal print in order to catch other problems, but I'm less confident that it wouldn't be falsely triggered during normal printing. At the very least that would need to be an option that can be turned off.

yonkiman, 2017-03-15:
I was just looking through all the different gcodes and discovered M913, which sets the motor currents to be a percentage of normal.  I'm going to add this to all my homing code so if/when there's a head crash/other collision the force will be significantly reduced.

Not exactly what you're asking for (which I agree is a good idea) but it might help to save other people and their printers from similar fates...

dc42, 2017-03-15:
I added the M913 command specifically so that current can be reduced during homing and/or bed probing moves.

I don't think keeping the bed probe active during homing would be helpful to most users. On Cartesian and CoreXY printers, it's common to use the Z probe to do Z homing anyway. Where a separate microswitch is used for low-end Z probing, the Z probe will normally trigger before the microswitch anyway.

During printing, the LED on my IR Z probe flashes on and off constantly.

closed 2017-03-15 by dc42