Duet3D/RepRapFirmware #33: Object height and layer times not being calculated

y00rek, 2016-02-23:
Hi,
I am running 1.09r-dc42 version on my Duet 0.8.5. Ever since I started commissioning my new delta (based on David's blog) I had problems with object height and other information about the file. I guess is must be my gcode. I use Slic3r 1.2.9. In G-code Files the object height is never shown. In print status Layer Time and Last Layer Time are showing as n/a. Layer statistics graph is never updated.

I attach the start.gcode(txt) (first 5 layers) and end.gcode(txt) (last 4 layers) of the file that gives me problems. Actually all my g-code files give me this problem. I had a quick look at the code and it looks like you are searching backwards for 'G1 Z' pattern that is not in the comment. In vi, when I do backward search I do find this pattern without any problems. 

Is this a bug or is it something in my generated g-code files?

[start.txt](https://github.com/dc42/RepRapFirmware/files/143391/start.txt)
[end.txt](https://github.com/dc42/RepRapFirmware/files/143392/end.txt)

pkmcnc, 2016-02-23:
I second that.
First I thought I was using G1 to lift the nozzle after a print, but with G0 it still shows the print height wrong.

dc42, 2016-02-23:
I'll try to take a look at this next week.

chrishamm, 2016-03-02:
I have a fix for this working and will include it in my next firmware release.

y00rek, 2016-03-02:
Is this fix included in 1.10? Currently I'm on 1.09 (both HTML & JS).

chrishamm, 2016-03-02:
This bug is not related to the web interface, it was an issue in the firmware which I have fixed in my fork.

y00rek, 2016-03-04:
David, when will you be able to address this? Any time frame?

y00rek, 2016-03-12:
Happy to report that the object height, layer times and the graph are now working.
Thanks guys :-)

y00rek, 2016-03-13:
Too quick with the final verdict :-(
Some files do show the correct object height, some do not.
I uploaded the file that gives me problems with object height. The layer times and graph are being shown correctly. Here is a link to a zipped file -
[http://www.yoorek.com/CandyBowl-DD-PLA-WHI-DELTA-STANDARD-L0.2-P5-B4-T0-I0%25.gcode.zip](http://www.yoorek.com/CandyBowl-DD-PLA-WHI-DELTA-STANDARD-L0.2-P5-B4-T0-I0%25.gcode.zip)

dc42, 2016-03-13:
Thanks for your example. The code that looks for the object height scans the last 128Kb of the file for the final G1 Z to find the object height. That particular file has a very complicated final layer, so the final G1 Z command is 148Kb from the end of the file. I will increase the size it scans in a future release.

y00rek, 2016-03-13:
Thanks. I knew there must be a logical explanation.

dposea, 2016-03-14:
if not found why not load the previous block instead of  hard coding a size. I know you thought of this already, and maybe it's too much work for the occasional problem file.

pkmcnc, 2016-03-15:
The problem remains with 1.09x.
Here is an example file https://www.dropbox.com/s/fzh6q0ookmb1vdo/podst.gcode?dl=0
G1 Z is 37KB from the end

chrishamm, 2016-03-15:
Thanks for that file. The firmware does parse the last G1 .. Z.. code, but since it's a code in your end G-code, it confuses the firmware. I've noticed one more issue which causes following ;End comments to be ignored, but I'll fix that in my next release.

Until then you can work-around this problem by creating a new macro file (e.g. LiftZ.g), adding your last move to it (i.e. G0 X0 Y0 Z700 F6000) and calling it from the end G-code (M98 PLiftZ.g).

y00rek, 2016-04-15:
:-( I guess I tend to print complex files. In this file [(sample.gcode.zip)](https://www.yoorek.com/sample.gcode.zip) the object height is not calculated as the last G1 Z is about 362KB from the end of the file. Furthermore when I try to list the folder containing this and some other files of similar size in DWC I get disconnected due to timeout error.
Would it be possible to have a fix for such scenario. I cannot select this file for printing from DWC and when using PanelDue I find I have to restart the printer and select the file. Once selected and printed subsequent selection for printing is not possible as the request is not acknowledged by PanelDue.

dc42, 2016-05-09:
I believe the issue with DWC disconnecting due to the firmware taking excessive time to find the file info is fixed in release 1.12a.

dc42, 2016-05-10:
Fixed in 1.12a.

closed 2016-05-10 by dc42