Duet3D/RepRapFirmware #35: Source files missing from lastest restructuring?

dcnewman, 2016-03-09:
SamNonDuePin.h no longer defines the extended pins X0, X1, X5, X7, etc.  However, they are still used.

Moreover, their definition does not appear in any header file.  Here's a search across a fresh copy of the dev repo,

% find . -name \*.h\* | xargs -n1 grep -l X1
./Libraries/EMAC/mini_ip.h
./Libraries/Lwip/lwip/src/sam/include/arch/cc.h
./src/Pins_duet.h
./src/Platform.h 

% grep X1 Libraries/EMAC/mini_ip.h
/*  0X14 through 0x1D Reserved for robustness experiment */

% grep X1 Libraries/Lwip/lwip/src/sam/include/arch/cc.h
#define X16_F           "hx"

% grep X1 src/Pins_duet.h 
const Pin ENABLE_PINS[DRIVES] = { 29, 27, X1, X0, 37, X8, 50, 47, X13 };
const Pin STEP_PINS[DRIVES] = { 14, 25, 5, X2, 41, 39, X4, 49, X10 };
const Pin DIRECTION_PINS[DRIVES] = { 15, 26, 4, X3, 35, 53, 51, 48, X11 };
const Pin HEAT_ON_PINS[HEATERS] = { 6, X5, X7, 7, 8, 9, -1 };			// Heater Channel 7 (pin X17) is shared with Fan1. Only define 1 or the other
const Pin Z_PROBE_MOD_PIN07 = X12;										// Digital pin number to turn the IR LED on (high) or off (low) on Duet v0.7 and v0.8.5 (PC10)
const Pin COOLING_FAN_PINS[NUM_FANS] = { X6, X17 };						// Pin D34 is PWM capable but not an Arduino PWM pin - use X6 instead

% grep X1 src/Platform.h
	//       initialized until configured via a "M305 Pn X10m" command with 0 <= n < HEATERS

dcnewman, 2016-03-09:
I see that the revised .project file refers to a directory

workspace_loc:/RepRapFirmware/arduino/variant

however that directory, arduino/, does not exist in the repo. Not sure if that is the missing info or not.

dc42, 2016-03-10:
The build system has changed with the new files. See the new BuildInstructions.txt file I have just added.

closed 2016-03-10 by dc42