Duet3D/RepRapFirmware #36: Sensor LED stopped working after update to 1.09x-beta3

y00rek, 2016-03-12:
After updating to 1.09x-beta3 (why no dc42 in the name btw) I noticed that the LED on the sensor board is not working during G32.
It seems to work otherwise. After the machine is powered up it blinks. When I put a piece of paper close to the sensor board the LED will light up as expected. During print it does blink as before. Only during the calibration (G32) it remains off.
Is this intentional? I thing it should be blinking when it approaches the bed. The calibration is being done as before. I know this is beta version, but so far no-one has reported this (AFAIK).

criticallimit, 2016-03-12:
Same here.
My reading when probe is high in the air is 85, not 0 as before.

dc42, 2016-03-13:
Thanks for your feedback. I confirm this is a problem. I will fix it in beta4.

dc42, 2016-03-13:
Fixed in 1.09x release.

closed 2016-03-13 by dc42