Duet3D/RepRapFirmware #37: M204 not supported, used by slic3r

dposea, 2016-03-14:
I tried to find info on this with google, I know it must have been addressed before, but I could not find anything.

dcnewman, 2016-03-14:
IIRC, this M204 usage was added to slic3r in an attempt to make up for firmwares which lack any form of pressure advance (e.g., Marlin).  That in turn was motivated by experiments of some users suggesting that with their printer and firmware, adjusting the maximum accelerations on a per line segment basis provided some benefit.  However, for firmwares which do have functional pressure advance mechanisms (e.g., RepRapFirmware) you should simply disable slic3r's use of M204.

nebbian, 2017-02-06:
Note, you can use a post processing script to turn M204's into M201's.  An example is here:
https://github.com/prusa3d/Slic3r/issues/106

dc42, 2017-02-06:
M204 is on the list for supporting in a future release of RRF.

dc42, 2017-04-27:
M204 is implemented in RRF 1.18.

closed 2017-04-27 by dc42