Duet3D/RepRapFirmware #38: M569 Support for axis indexes

pholmes2012, 2016-03-14:
Are there any plans to add support for multiple drives per axis?
I am asking because the Duet only has one socket for the Z-axis which often requires 2, if you don't want to solder 2 motor leads together.

I will be implementing this locally.

dc42, 2016-03-15:
This is on the list of things that may be implemented in future, but it isn't a priority. Most users don't want to use up a second driver when one driver can easily drive two motors in series. It's easy enough to make a splitter cable. The next version of the Duet will probably have two Z motor connectors.

pholmes2012, 2016-03-15:
Ok no problem, I have now implemented this functionality here, two per axis anyway. It does add a few microseconds to the ISR but apart from that it is working, so if you do decide it is worth implementing in the future then please let me know.

T3P3, 2016-04-07:
This could also allow heavier axis to be driven by multiple stepper drivers (for example a massive heated bed using two Y axis motors -  a request I have had).

pholmes2012 can you share your modifications please?

nafets77, 2016-04-08:
I would like to use E1 to drive my second Y-axis motor on a printer with a a beefy gantry, modified Shapeoko2, and your solution is of great interest pholmes2012.

dc42- the ability to independently decide if you want to use a spare driver for dual Z, dual Y etc. would be a great feature on a future release.

pholmes2012, 2016-04-08:
I will get the code changes posted this weekend, though I am a software
engineer by trade I am new to github and am not currently sure how I get
the code changes there without actually checking them in to the trunk, any
advice would be appreciated.

dc42, 2016-04-08:
This facility is on my list to add to RepRapFirmware. However, it's low priority, because connecting the two motors in series is generally an entirely adequate solution, especially if using 24V power.

T3P3, 2016-04-08:
pholmes, that would be great, best plan is if you fork DC42's version of the firmware on github and then change your fork.

paenian, 2016-05-02:
I thought about using three steppers on Z, as a way to provide bed leveling on a large printer.  If you could selectively disable two of them, you could flatten the bed manually, instead of angling the base of the part a bit.

Also wondering if I can drive three motors with one driver as an interim.

dc42, 2016-05-02:
Driving 3 motors in series from one driver should be no problem if you use 24V power. Using 3 independently-driven motors to do bed levelling is on my todo list.

Jmjcoke, 2016-06-23:
I would say this is the biggest reason I have shelved the Duet for my current build, l like and appreciate many of the features and bought the ir height sensor, 7" paneldue and the controller with expansion card thinking axis mapping would be simple and since this was the reason I switched to repetier FW years ago I never thought the next generation of FW wouldn't feature one of the most useful way to make use of 9 stepper dives never crossed my mind. What a complete wast of money and time. How many freakin extruders do you need? Oh well I gues l will put it on my delta and use the X3 pro on the big machine. I will wait until the smoothieboard v2 pro comes out, because they do not care how many drive I want to be X axis.

dc42, 2016-06-23:
Don't be absurd. Connect your Z stepper motors in series, it's very to do and saves using up stepper driver channels. Or talk to pholmes2012 and ask for his firmware patch. Or talk to digital_dentist on the RepRap forums, who will tell you why you shouldn't use multiple Z motors anyway.

I'll add support for driving multiple stepper drivers in parallel at some time, but I have more important things to implement first that don't have easy workarounds. But hey, RepRapFirmware is open source, so anyone is free to add the features they want.

Jmjcoke, 2016-06-23:
Not supposed to run two stepper mot on Z axis? My Z axis is 38" tall with
40" between the acme rods, I would love to hear the convoluted method of
driving it with one motor. Well the machine is running great on the X3pro,
2 X axis drives, 2 Y axis dives, 2 Z axis drives and 2 extruders, the
drives get hot like that so running on single stepper per axis is not going
to work, so I will keep being absurd. I blame myself, in the Repetier FW I
always saw due as an option and thought it would support the most used due
based boards, obviously a couple hundred dollar mistake.

dc42, 2016-06-24:
Just listen and CONNECT YOUR TWO STEPPER MOTORS IN SERIES. That won't make the drivers hot, unlike connecting them in parallel. There is just one potential disadvantage, which is that if you are only using 12V power then the torque at high speeds may be reduced. Whether this happens or matters will depend on your motors, your steps/mm, the mass of the moving parts, and the top speeds you use. You haven't told us any of those, so I can't make a prediction.

jamesarm97, 2016-06-24:
@Jmjcoke It can be done. My 300x200 corexy uses one Z stepper, my 330x330x480 corexy uses 1 Z stepper, Jetguy's 24" and 1 meter corexy use 1 Z stepper. Just takes a few pulleys and a belt.

Jmjcoke, 2016-06-24:
OK, I have never connected a stepper in series. I read about how it works
and will try it. the X axis is being operated as a dual carriage, once that
is functional in the Duet FW I will try again.

Jmjcoke, 2016-06-24:
I am using 24V.

dc42, 2016-06-24:
Jmgcoke, do you mean two separate X carriages with one print head on each, where you park them on opposite ends of the X axis when not in use?

paenian, 2016-06-24:
I have three steppers connected in series - I made a jumper cable to do the
splitting, so I don't have to modify the stepper cables at all.  One end is
the Duet connector, the other is three stepper connectors:

Jmjcoke, 2016-06-24:
I know anything can be done mechanically, but that takes time and money, I
already had 2 openbuilds c beam actuators 1000mm long with N23 motors,
right now it is much easier to click with my mouse in repetier config tool
and select mirrored Z motors and select which driver I would like it to be.
There are multiple things that I really like about dc42's firmware and I
will implement on a machine in the future, this one has the dual x carriage
system so once that is a working  option in this FW I will try the series
wiring.

Jmjcoke, 2016-06-24:
Yes, 2 separate carriages that park.

Jmjcoke, 2016-06-24:
Repetier just got this working in the dev fork.

Jmjcoke, 2016-06-24:
The cable sounds like the way to go and thank you for the information, my
azteeg X5 just found a new home.

dc42, 2016-06-24:
> Yes, 2 separate carriages that park.

Thanks, now that I know another person is waiting for that, I'll move it up the priority list. I'll quite likely support multiple drivers per axis at the same time, because the configuration code needed will be common to both.

dc42, 2016-06-28:
I am.about to start on some related work, so I may be adding support for multiple motors per axis soon. I have already defined a new gcode  (M584) to configure this. What is a reasonable maximum for the number of motors on an axis? How about 4?

dcnewman, 2016-06-28:
FWIW, the people I know making "large" Core-XY & Cartesian printers end up using four leadscrews so four is not unreasonable.  Four supports are used rather than three owing to the sag of the build plate when it's 1m on a side.  (However, the people I know doing these only need one NEMA 17 or 23 stepper motor and slave all four screws with a timing belt.  The mechanical advantage of a TR8x8 leadscrew is considerable, after all.)

Jmjcoke, 2016-06-28:
I think 4 is great.

eboston, 2016-06-28:
You know, based on Murphy's Laws, the number needed will always be 1 more than the number you choose.  :)

dc42, 2016-06-29:
If the order in which the driver numbers are specified didn't matter then I wouldn't need to set a limit because I could use a bitmap to store the driver numbers used for each axis. However, for multiple X carriages and for triple Z motors that are also used for bed levelling, the order does matter.

I'll allow up to 4 motors per axis. The Duet WiFi only supports 10 drivers anyway, so it's unlikely that you would want more than 4 assigned to a single axis.

paenian, 2016-06-29:
I'll try to get my dev/build environment up and be ready to test/code this
weekend.  Thanks for all your work on the firmware!

My doppelbot triple-z-motor machine is up and printing well with the Z
motors wired in series; I'm assembling a refined version now.

Thanks again,
Paul

dc42, 2016-06-29:
Just to be clear, initially I will be supporting **synchronised** movements of more than one motor per axis only, because that fits in with work I am doing on the step pulse generation. But I am designing the framework to allow support for independent X carriages and independently-driven Z motors for bed levelling to be added without too much disruption. That's why I want to preserve the order of the driver assignments.

We haven't mentioned extruders. I am assuming that there is no requirement to assign more than one driver to each extruder.

paenian, 2016-06-29:
Gotcha, David - that's why I wanted to get my build up so I can contribute
:-)

My day job is programming, which makes me feel guilty - I'm able to write
code, but generally not enthusiastic about it by the time I'm off work.

I can't think of any reason to dual-drive extruders.

Paul

dc42, 2016-07-02:
Multiple drivers (up to 4) for a single axis are supported as of version 1.14. There is a new command M584 to configure which drivers to use for each axis. Please note, any M360 and M906 commands in your config.g file should come after the M584 command(s) so that they apply to the correct motors, and so that all motors for a single axis use the same microstepping and current.

I will close this issue if nobody reports any problems with this feature in the near future.

Jmjcoke, 2016-07-02:
Great! Is there a place to donate?

closed 2016-12-20 by dc42