Duet3D/RepRapFirmware #39: Porting to STM32F2XX based controller

Hotaman, 2016-03-17:
dc42,

Your build instructions say to build the CoreDuet project first in the same workspace. What is the purpose?

I'm just getting started on the port so any tips would be greatly appreciated. I plan to build your code as-is this weekend before I start hacking on it. The controller is going in a Kossel Mini and I like the way you are handling the Deltas so I would like to use your firmware on my hacked RAMPs controller. I'm just using up some bits I found laying around on my bench to get ARM horsepower for my new Delta.

 I don't want to distract you away from this awesome work. If you have no interest in my project just close this issue and I won't bother you any more. I'll figure it out, but I'll do it faster with a little help TIA.

dc42, 2016-03-17:
The CoreDuet project replaces the patched Arduino core that was used by earlier versions of my fork of RepRapFirmware. The intention is that it will eventually contain all of the code that interfaces to the hardware. That will make porting RRF to new processors easier. I have work in progress on a new Core project for the next generation Duet.

When  you bud CoreDuet, it creates a library that is used in the link stage when you build the RepRapFirmware project.

Hotaman, 2016-03-18:
Thank you for the vision!

I'll start my port with that library this weekend after completing hardware tests.

The Duet is a Very nice controller. I would go with it but it is currently out of my price range for this build. If what I'm working on pans out, it will be a minimalist ARM based controller you can build in a day for less than $50. I'll start a repo for it once I'm sure the hardware is solid.

dc42, 2016-03-18:
You might like to take a look at the minimalist controller project I started at https://github.com/dc42/Basic-3D-Printer-controller. The schematic capture is complete apart from the expansion connector, but the PCB layout needs more work. It uses an Atmel SAM3S processor, so porting RepRapFirmware to it would be easy.

Hotaman, 2016-05-31:
@dc42 time flies and situations change. Work consumed all my spare time so I have money and no time now. As a result I acquired a Duet board for use on a Kossel Midi I built.

I'd like to use my IR Z-probe but it uses a 3.3v hobby servo to raise and lower the probe end. It would appear that RRP does not support hobby servos which I find pretty odd at this point in time since they are so darn useful.

Do you have a planned path to add servo support?
Best approach?
Has anyone added servo support yet?

Is there any docs or examples of adding a new GCode to RRP?

I don't want to make a new wheel unless I have to.

dc42, 2016-05-31:
Until this week, nobody had asked for servo support in RRF. With so many types of contactless Z probe available, along with solutions that use the nozzle itself as the probe, using a servo to deploy the probe seems to used only in older printer designs. However, I'll consider supporting M280 or M340. It will probably be necessary to hijack either one of the additional heater channels or the second fan channel, because all the PWM channels on the microcontroller are already allocated to heaters and fans.

Hotaman, 2016-06-02:
That's what I figured, I haven't really seen it mentioned anywhere other than it's not available. I've been using my design for a few years now and it has been rock solid so I would like to keep using it. I finished getting the Delta setup on the Duet today and the first two prints were near perfect!

I haven't explored the non-contact designs much. How does your modulated IR sensor handle a glass bed that may have a variable amount of transparency due to glue film?

For the servo I was looking at adding M280 and using the last extruders heater PWM since it would probably be the least used PWM channel. Adding the gcode looks pretty straight forward, I'm just not sure what needs to be done to wrestle the pin away from the heater.

Hotaman, 2016-06-05:
After realizing the last heater is shared with Fan1 it's probably not a good choice. I'm closing this issue and I will open a new one for servo support to keep things sane and organized.

closed 2016-06-05 by Hotaman