Duet3D/RepRapFirmware #40: capability to control the ready to use cutting modules such as auto focus laser head or plasma THC

mkeyno, 2016-03-22:
hi , I was wondering if this firmware have capability to control the ready to use cutting modules such as auto focus laser head or plasma THC , 
I saw couple of effort on Marlin enthusiast to check output sensor the cutting tool and control the stepper motor the module to control height during the printing 
any comments really appreciated

dc42, 2016-06-16:
Have you looked at the M571 gcode, http://reprap.org/wiki/G-code#M571:_Set_output_on_extrude ? That is already being used to control at least one laser engraver. The FAN0 output is re-purposed when you use M571.

mkeyno, 2016-06-16:
thanks @dc42  do you know link or repo which is already use for such purpose

dc42, 2016-06-16:
I'm not sure exactly what you are looking for.The M572 gcode facility has been supported by the standard firmware for a long time. It turns on the FAN0 output with the specified PWM ratio whenever the printer is extruding. So by using it to control a laser, you have a laser engraver. You mentioned controlling the height, but I don't know exactly what you want.

mkeyno, 2016-06-16:
sorry if I explain not good , what I  looking for is the code implementation(G codes) or firmware implementation of closed feedback loop for height sensor measuring  and adjustments of  Z-axis for laser or plasma tools, but during posting  this ticket , I find some guys in Marlin and Smoothieware follower implemented such feature , I was wonder if yet any attempt has been made for such feature in RepRapFirmware

dc42, 2016-06-16:
No, we haven't implemented that in RRF. What type of height sensor is normally used in this application?

mkeyno, 2016-06-16:
In metal cutting laser and plasma CNC , used the capacitor  sensor but I find the [precise short range laser measurement ](https://www.ia.omron.com/products/family/3175/)  which is good for both metal and non metal cutting .
Normal cutting head consist of  laser head , stepper motor and controller, ranged form 1000$ to 3000 $ from Chines supplier , with this feature and this sensor I could cut down more than half for THC CNC machines

T3P3, 2016-06-17:
This just popped up in my news feed:
http://hackaday.com/2016/06/05/in-soviet-russia-diy-laser-rangefinder-scan-you/
$18 or less for a laser rangefinder hack that although is for a different application here might work for what you are talking about. how precisely does the surface need to be followed in this sort of laser cutter?

mkeyno, 2016-06-17:
thanks @T3P3 Tony for link , really interesting although it is for scan purpose but I'm sure similar approach for single range finder wont be hard , however [following is the ticket discuss](https://github.com/MarlinFirmware/Marlin/issues/3208#issuecomment-213214199) about this feature and today couple of feature add it to Marlin , but as I never use the RepRap yet , I don't know where should start for such feature

dc42, 2016-12-20:
Support for a laser toolhead is on the work list. From the thread you linked to, the main requirements appear to be:

- G0 and G1 aren't the same anymore. G0 move the head without firing laser, G1 with firing laser. Firing laser is also with G2 and G3 arc moves.
- G7 to execute a raster line ( base64 encoded line )
- M3 start laser at specific intensity/power
- M4 Turn on laser
- M5 Turn off laser
- M649 Set laser options

Anything else

mkeyno, 2016-12-27:
thanks @dc42  but I was aware for such feature for many cnc firmware, but in actual  printer need high priority function to bypass G-code for safety and real time control like what is implemented in GRBL and smoothee , for high energetic cnc tools this essential and I could successfully install  GRBL  on my high power laser cnc machine

dc42, 2017-03-15:
Are you talking about an emergency stop feature that doesn't use gcode, or something else?

mkeyno, 2017-03-17:
yes , actually there are features in GRBL which let the user to control job in unexpected circumstances , those have higher priority than Gcode, I mean send the g-code and waiting in ques to process is really unwise action in unexpected events , does RepRapFirmware has such feature ?

raykholo, 2017-06-22:
@dc42 Did any progress occur on this development? 

I wanted to explain how we are doing it with the Grbl/ Smoothie and LaserWeb at the moment in the hopes that it helps you:

M3 is a straight shot laser turns on, and M5 is a straight turn off.   In the CNC world, M3 does also have the S parameter to set the spindle speed... 

M4 showed up in grbl, please see here and search for the 3rd instance of "M4" on the page: https://github.com/gnea/grbl/wiki
"Dynamic Laser Power Scaling with Speed"

I would suggest leaving M4 out, and recognizing that just M3 is full power on while M3 Sxxx will go at a proportional power level.  This gets a bit complicated because the official cnc gcode spec calls for just an M3 with no parameters to resume the previous speed from the last M3 command that had an S word, etc...

As you correctly stated, G0 and G1 are different.  G0 is a travel move while G1 is a fire move. 

The format Smoothie/ Grbl/ LaserWeb uses is G1 X10 Y10 S0.4 F600 where S is the fire level ranging 0 - 1 in this case.  That is really it.

G7 - is this something like streaming pixels directly for raster?  This has been a controversial subject between several firmware teams for some time now... Presently high throughput rasters are still being done over the very high quantity of G1 line segments.  

M649 - what settings do you have in mind? 

With no disrespect intended to anyone, I would not use Marlin as an example of how to do laser streaming.  Right now GRBL and Smoothie are what most people are using with LaserWeb4 using primarily the G0 and G1 lines.  

In addition to the wiki link above, this one about the grbl laser mode explains a lot: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode
This explains the CNC vs Laser nuances, 

Cheers,
Ray Kholodovsky 
Cohesion3D

T3P3, 2023-05-04:
@raykholo I am going though and housekeeping old issues.

There have been significant additions to RRF since your input on this issue:

In laser mode (see[ M452](https://docs.duet3d.com/User_manual/Reference/Gcodes/M452))
G0 is a move w/o laser firing
G1 Snnn ; nnn = laser power

(from RRF 3.5.0beta3_ In laser mode a G1 command may now have up to eight S parameters to specify the laser power at uniform intervals along the move (raster clustering)  (So the request for G7 is probably redundant)

M3 Snnn ;laser power (laser mode). In laser mode, the relationship between the S parameter and laser power depends on the R parameter that was specified in the M452 command.

In addition there is a specific feature request to allow for a low power mode for laser focus:
https://github.com/Duet3D/RepRapFirmware/issues/697

I would be interested if there are any outstanding requirements for Laser support in RRF?

T3P3, 2024-03-06:
please reopen if still relevant