Duet3D/RepRapFirmware #41: [Request] Velocity change smoothing configuration (ie. jerk parameter).

mirage335, 2016-04-05:
Some kind of "jerk" parameter, adjusting the smoothing between velocity changes, may improve print quality in some cases.

Theoretically, this might be implemented as a limit on the rate of change in acceleration (ie. jerk), just as acceleration is the rate of change in speed.

For TazMega, the threaded rod drive can couple extreme instantaneous force into the workpiece and toolheads. Even with 24 V-Slot wheels, this seems to cause the bed to bounce vertically when suddenly reversing direction. This may be contributing to front and back areas of printed parts showing gaps, as well as rough layers at heights above two centimeters.

Please consider adding some more tuning knobs to manage acceleration parameters.

dc42, 2016-04-05:
This is already implemented, via M566.

mirage335, 2016-04-05:
Awesome, much thanks!

closed 2016-04-05 by mirage335