Duet3D/RepRapFirmware #42: M36 destroys Telnet

tomasf, 2016-04-05:
M36 doesn't seem to work from a Telnet connection. There's no response, and Telnet commands stop working completely until I restart the Duet board. The same command works fine in the web G-code console.

M36 /gcodes/foobar.gcode

Firmware version: 1.10+4 (2016-03-30)

y00rek, 2016-04-05:
I believe there are more commands behaving like this. G30 S-1 is one of them. After using it in telnet session I can no longer use the connection. I have to kill telnet process.

dc42, 2016-04-07:
Thank you both for your reports. I am aware that there was a problem in this area with some client programs in some earlier versions of the firmware, but this was believed to be fixed in version 1.10+4.

y00rek: have you observed this behaviour using version 1.10+4, or only in earlier firmware versions?
Both of you: what Telnet client programs are you using, and under what operating system?

tomasf, 2016-04-07:
I'm using the built-in telnet client in OS X 10.11. I also tried Prompt on iOS and got the same behavior. Disconnecting and re-connecting does not fix it. Even connecting from another machine doesn't work. Rebooting the Duet is the only thing that seems to help.

I just noticed the web interface becomes very hard to load after crashing telnet. I randomly get a 404 error for some included JavaScript file. I reload and get 404 on some other file. I try a few times and finally get past it loading initially, but then some XHR call fails instead. I wasn't sure this was related to the telnet thing at first, but I can reproduce it every time by using the M36 code. Strange.

y00rek, 2016-04-08:
@dc42 I am using native telnet client that comes with OS X (10.11.4). I am experiencing this problem with the latest version of the firmware - _FIRMWARE_NAME: RepRapFirmware FIRMWARE_VERSION: 1.10+4 ELECTRONICS: Duet 0.85 DATE: 2016-03-30_

The steps I perform to get the problem are -

- G1 Z10 F8000
- G1 Z5 F250
- G30 S-1

At this point the telnet session is dead (as in I cannot issue any other commands like M115 or G28 as there is no response). These commands work in DWC however. The only command I can execute is exit which will terminate the telnet connection and return me to shell prompt. Starting new telnet connection does not help. Emergency stop does not help either. I have to cycle the power to Duet.

chrishamm, 2016-04-08:
I have greatly improved the Telnet feature in my latest firmware release, so it might be worth a try to check if that resolves your problem(s). You can find a precompiled binary here: https://github.com/chrishamm/RepRapFirmware/blob/dev/Release/RepRapFirmware-1.11-ch.bin

y00rek, 2016-04-08:
I guess I have to wait for dc42 version to include these changes.

y00rek, 2016-04-08:
@dc42  & @chrishamm - Is there any difference between dc42 & ch versions? I religiously stick with dc42 version as I believe it has some features not available in the other.

@dc42 - I remember reading somewhere that there is a problem when extruder elasticity, z-lift and firmware retraction together with axis compensation is being used. Is this still the case in the latest version (1.10+4)? The reason I am asking is that if I'm using firmware retraction the first layer is always too thin no matter what I do with probe trigger height.

tomasf, 2016-04-08:
@chrishamm Nice, M36 works fine over Telnet in your 1.11. Thanks!

chrishamm, 2016-04-08:
@tomasf, that's good news, thanks for testing it!
@y00rek, my fork is based on an older version of dc42's firmware and since I started development I've merged in almost all of his changes (and dc42 has merged in many of my own improvements), so I think basically every code that is supported by dc42's fork is supported by my fork as well. There are a few minor differences, but the overall functionality should be fairly equal.

I had a chance to test dc42's fix the other day and I can confirm it's working well in my fork.

dc42, 2016-04-09:
yoorek, the fix for pressure advance, Z lift and firmware retraction is included in 1.10+4. I am working on a new release, which will include chrishamm's telnet change, and I expect to have this ready at least in beta form later today. But try chrishamm's fork if you want, it should be up to date with my fork in the most important respects.

tomasf, 2016-06-05:
This seems to be fixed in 1.12. Thanks! Closing.

closed 2016-06-05 by tomasf