Duet3D/RepRapFirmware #43: M207 problems

y00rek, 2016-04-10:
I believe there is still a problem with M207. When I had everything properly calibrated with M207 turned off the first layer height was being set correctly after G32. As soon as I enabled firmware retraction (M207) the first layer is very thin. Either I am doing something wrong or there is a bug there somewhere.

BTW how does one set the IR probe trigger height? Before calibration or after calibration? I do it after running G32. I lower the nozzle until it just grips the paper. Then -
G92 Z0
G1 Z5
G30 S-1
G1 Z5
G30 S-1
G1 Z5
G30 S-1
Then I average out the results and enter it in config.g G31. I restart the printer and run G32 again.
I also run G32 (6 factor, 13 points) before every print.

dc42, 2016-04-11:
Thank you for your report. Please can you check and report the following:

1. Does it make any difference if you turn off Z hop in the M207 settings?

2. Are you using absolute or relative extruder coordinates in your gcode file?

3. Are you using pressure advance?

4. When you print the same model sliced with and without firmware retraction, is the Z height displayed while printing the first layer correct in both cases?

y00rek, 2016-04-12:
Sorry for small delay with the answer. Here is what I found (using 1.11-beta1 (2016-04-10))

1) Have not checked this one yet.
2) I'm using relative extruder coordinates
3) No, I'm not using pressure advance in slic3r (it is set to 0)
4) The height is displayed correctly when printing a model with no firmware retraction (Z = 0.10). Same model sliced with firmware retraction displays Z = -0.20 and the first layer is very thin.
5) I mean the layer height.

dc42, 2016-04-12:
What I think might be happening is that you are using Z hop (nonzero Z parameter in your M207 command) and you have a G11 un-retraction command in the start gcode. Please examine your gcode file to determine whether the first G11 code occurs before or after the first G10 code.

When I asked about pressure advance, I meant RepRapFirmware pressure advance, configured via M572.

y00rek, 2016-04-12:
Yes I'm using Z hop (M207 S8.0 R0.000 F4500 Z0.500).
I have pressure advance configured (M572 P0 S0.10)
In my code I have G10 first, then G11 (this is not in my start code however). Anyway here is the beginning of my code up until first skirt

```
; generated by Slic3r 1.2.9 on 2016-04-12 at 17:12:07

; Default print settings for my Delta

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.40mm
; infill extrusion width = 0.40mm
; solid infill extrusion width = 0.40mm
; top infill extrusion width = 0.40mm

M107 ; disable fan
;
; Custom start G-code (from slic3r)
;
M404 N1.75 D0.4 ; Set filament size and nozzle diameter
M300 S5000 P300 ; sound notification
T0 ; Select first extruder
M107 ; disable fan (this is deprecated in RepRap Firmware)
M106 P0 S0 ; disable cooling fan (first extruder)
M106 P1 S0 ; disable cooling fan (second extruder)
;
M190 S70 ; set bed temperature for first layer and wait until reached
M300 S5000 P300 ; sound notification
M104 S210 ; set extruder temperature for first layer and proceed
;
G32 ; Autocalibrate (do it after the bed has reached its temperature)
;
;
; Put a code to make the head go to its waiting position
G0 X-148.00 Y0.00 F8000
;
M109 S210 ; set extruder temperature for first layer and wait until reached
M300 S5000 P1000 ; sound notification
;
; Ready to print
;
G21 ; set units to millimetres
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;
; Start of layer 0
;
M300 S5000 P300 ; sound notification
G10 ; retract
G1 Z0.100 F15000.000 ; move to next layer (0)
G1 X-20.125 Y29.043 F15000.000 ; move to first skirt point
G11 ; unretract
G1 X-24.928 Y29.043 E0.13149 F2586.676 ; skirt
```

dc42, 2016-04-12:
I can see the problem. It's in this bit of the gcode:

G10 ; retract
G1 Z0.100 F15000.000 ; move to next layer (0)
G1 X-20.125 Y29.043 F15000.000 ; move to first skirt point
G11 ; unretract

There is a G1 Z0.100 after the G10 command to set the first layer height, but then the G11 undoes the z-hop, putting the head too low. Looks like your slicer doesn't allow for the possibility of your firmware retraction having a z hop. 

I guess the firmware could be changed to make the un-retraction conditional, but I'm not sure this could be made foolproof, for example if the print is paused and restarted. IMO it should be fixed in the slicer. What slicer are you using? I generated a file using slic3r 1.2.9 with firmware retraction enabled, and it generated correct code:

G1 Z0.200 F7800.000
G10 ; retract
G1 X32.408 Y1.281 F7800.000
G11 ; unretract
G1 X32.066 Y9.944 E0.25743 F1800.000

In this case it moves to the first layer height before retracting instead of after, so the un-retraction doesn't cause an incorrect Z.

y00rek, 2016-04-13:
I see. I'm using Slic3r 1.2.9 on a Mac. What are your settings in Slic3r? Mine are -
[image]
[image]

I will try to play with the settings to see if I can force Slic3r to generate g-code like you.

y00rek, 2016-04-13:
I was able to generate the first layer code like yours by turning off 'retract on layer change'. Do you have it turned off in yours?

```
;
; Start of layer 0
;
M300 S5000 P300 ; sound notification
G1 Z0.200 F15000.000 ; move to next layer (0)
G10 ; retract
G1 X17.000 Y-13.007 F15000.000 ; move to first skirt point
G11 ; unretract
G1 X18.791 Y-12.611 E0.09584 F5114.789 ; skirt
```

dc42, 2016-04-13:
No I don't have retract on layer change ticked.

I'll give some thought to tracking whether an un-retraction is pending and adjusting the Z height in any G1 commands when it is. The main issue is likely to be getting it to work properly when a print is paused and then resumed.

y00rek, 2016-04-14:
I'll leave the _retract on layer change_ turned off for the time being. I guess this is Slic3r's fault for generating g-code in this fashion.

dc42, 2016-12-20:
I've put a fix for this in firmware 1.17RC3 which I intend to release later today or tomorrow. Please test it with retract on layer change enabled.

dc42, 2017-03-15:
No feedback received so I am closing this.

closed 2017-03-15 by dc42