Duet3D/RepRapFirmware #44: Network interface crash

Aloderendra, 2016-04-30:
On firmware version 1.11 with no tool selected, requesting a print causes the network interface to crash/become unresponsive (Using Ormerod). 

Steps to reproduce (From web interface):
Request a print
Click pause
Click cancel (At this point, no tool is selected and the bed is Off)
Without re-enabling any tools, request another print.
The network interface is now shut down/crashed and will not recover without the power being cycled/reset.

The firmware still executes the print/queued commands just fine, but the network is now gone.

dc42, 2016-05-20:
I'm sorry, I can't reproduce that problem. However, I have a report of what may be a related problem. That user reports that if he pauses a print, then disconnects the web interface, he is unable to reconnect. I can't reproduce that either. Do you see the same problem? What commands do you have in your pause.g file?

Aloderendra, 2016-05-21:
If after I pause & cancel a print, I click to enable a tool - it works just fine. 

My pause.g is fairly straight forward (And unmodified from the default as far as I know): 

; Pause macro file
M83					; relative extruder moves
G1 E-4 F2500		; retract 4mm
G91					; relative moves
G1 Z5 F5000			; raise nozzle 2mm
G90					; absolute moves
G1 X0 Y200 F5000	; move head out of the way of the print

rwg0, 2016-06-27:
I see the same issue - to provide some more detail

* Pause and cancel a print - this puts both heaters to 'Off'
* Start another print - the bed header goes to active, but Heater 1 stays 'Off' although it gets a target temperature
* The bed warms up to target value and then calibration starts
*The hot end is still cold, so no extrusion happens, but the print starts in spite of an M109 instruction in the GCODE
* The printer prints a little of the first layer then goes crazy - this may involve homing the axes or moving seemingly randomly. At this point the network interface drops out (evidence - AJAX communication failure).

I suspect that the network crash is just a symptom and that the real issue is related to the hot end not being re-enabled and managing to somehow get past the M109.

cheers,
Robin

Start of GCODE file being used:

```
; generated by Slic3r 1.2.9 on 2016-06-23 at 22:07:44

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.48mm
; infill extrusion width = 0.48mm
; solid infill extrusion width = 0.48mm
; top infill extrusion width = 0.48mm

M107
M190 S65 ; set bed temperature
M104 S200 ; Start heating hot end
G28 ; home all axes
G32 ; Autocalibrate - disable with ; if not required
G1 X0 Y0 Z5 F8000 ; centre nozzle 5mm above bed
M109 S200 ; Wait for hot end to reach temperature

G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 E-3.00000 F2400.00000
G92 E0
G1 Z0.400 F7800.000
G1 X18.668 Y0.000 F7800.000
G1 E3.00000 F2400.00000
G1 X18.668 Y15.000 E3.83356 F2520.000
G1 X18.274 Y16.655 E3.92811
G1 X17.175 Y17.954 E4.02266
G1 X15.000 Y18.668 E4.14988
```

dc42, 2016-12-20:
Please re-test using firmware 1.17RC1 or later.

rwg0, 2016-12-20:
With 1.17RC2 the heaters remain active after the pause and cancel (not sure if this is intentional or not?). Anyway, all works fine apart from that - you can pause and cancel a print and then print something else correctly without any issues.

thanks!
Robin

closed 2017-03-15 by dc42