Duet3D/RepRapFirmware #45: [bug] G1 commands with endstop options are ignored

natewalck, 2016-05-14:
It appears that G-code commands that contain an S1 directive are ignored entirely.

For instance, if I run this G-code:

```
G1 X240 F3000 S1
```

Absolutely nothing happens.

If I take off the S1, like so:

```
G1 X240 F3000
```

It moves X 240 at a rate of F3000 as expected.

This of course makes homing impossible as it requires those commands to stop once hitting the endstop switches. In effect, if I run a homex.g script which looks like this:

```
G91                ; relative mode
G1 X240 F3000 S1   ; move up to 240mm in the +X direction, stopping if the homing switch is triggered
G1 X-4 F600        ; move slowly 4mm in the -X direction
G1 X10 S1          ; move slowly 10mm in the +X direction, stopping at the homing switch
G90                ; back to absolute mode
```

It effectively runs this:

```
G91                ; relative mode
G1 X-4 F600        ; move slowly 4mm in the -X direction
G90                ; back to absolute mode
```

Let me know if you need any additional information.

dcnewman, 2016-05-14:
I've yet to encounter endstop detection with G1 ... S1 not working, even with the most current sources.

What have you set for the endstop configuration with M574?  You might want to issue a M119 command and see if the firmware thinks that the endstop is triggered when it isn't.  You may need to put a M574 command into your sys/config.g file to establish the logic "sense" of the endstops as well as whether they are min or max for their respective axes.

natewalck, 2016-05-15:
@dcnewman Thanks! I'll take a look at that. I have my configs on github, hopefully that will help figure out what is going on: https://github.com/natewalck/TazStiffBlack-SDCard/blob/master/sys/config.g

natewalck, 2016-05-15:
```
M574
Endstop configuration: X high end (active low), Y high end (active low), Z low end (active low), E (active high)
```

```
M119
Endstops - X: at max stop, Y: at max stop, Z: at min stop, Z probe: not stopped
```

natewalck, 2016-05-15:
Looks like having S1 on the M574 command was causing the issue, testing it out now.

natewalck, 2016-05-15:
Ok, so if I set S1 on M574, it tries the homing command correctly, but in the wrong direction. If I slip it from X2 to X1, it still goes in the wrong direction.

natewalck, 2016-05-15:
Yep, this was user (n00b) error.

closed 2016-05-15 by natewalck