Duet3D/RepRapFirmware #46: Feature request: GCode file usage statistics

y00rek, 2016-05-17:
Hi David,

As I am using my newly built delta more and more I find that it would be useful to display alongside each uploaded gcode file its last printed time, number of times the file has been printed and actual print duration.
Would this be possible? I don't know if Duet has a notion of current date and time.

shadow651, 2016-05-17:
I would think you could at least store the number of times printed in
metadata or a sidecar file. I don't know if the printer knows the current
time after a power cycle, I suppose it could poll a computer though the web
interface.

HowlingMind, 2016-12-06:
There was a plugin for octoprint to do this that I used all the time (before upgrading to duetWiFi). It was very nice for repeat prints. The estimates are not always very good (and even the estimates aren't shown until you start to do a print), and knowing how long it took the last time (average if multiple times printed?) is really nice when trying to plan things out.

You don't really need to know the current date and time, just the total duration of the print (in seconds). The duet doesn't need to subtract start time from finish time. That being said, I haven't looked at the code to see if it tracks the time, and to see if there is an easy way to do a pseudo database to store the info.

brunofporto, 2018-08-03:
Also if it register the real printing time it can help future algorithms to give a better time prediction :D

dc42, 2018-08-03:
The actual or simulated print duration is already written to the end of the GCode file, and displayed by PanelDue.

T3P3, 2023-05-04:
Having an (optional) method of tracking file usage should be considered

T3P3, 2026-09-15:
Forum threads:
https://forum.duet3d.com/topic/35142
https://forum.duet3d.com/topic/33569
https://forum.duet3d.com/topic/33498
https://forum.duet3d.com/topic/12681