Duet3D/RepRapFirmware #47: M0 kills Telnet

tomasf, 2016-06-05:
When I use `M0` from Telnet, nothing happens. The print continues, I get no reply, and Telnet functionality doesn't work until I reboot the Duet. Even new connections are non-responsive; the TCP connection opens, but no G-code commands work.

The behavior is similar to #42.

tomasf, 2016-06-05:
After some additional investigation, it turns out that this only happens if M0/M1 is run during a non-paused print. If there isn't a print running or if the current print is paused, it works as it should and Telnet continues to work.

dc42, 2016-06-05:
Please try firmware 1.13 beta 1 and see if this still happens, also check it for issue #42 again.

tomasf, 2016-06-05:
Oh, nice. It is indeed fixed in 1.13b1. #42 is also (still) fixed. Thanks!

closed 2016-12-20 by dc42