Duet3D/RepRapFirmware #48: Simulation mode: G28 not simulated

tomasf, 2016-06-05:
When in simulation mode, `G28` seems to be ignored. Thus, if my delta hasn't been homed before simulation, I get the `Attempt to move the head of a delta printer before homing the towers` error.

I'm guessing `28` should be in the simulation code whitelist in `HandleGcode`.

Firmware Version:	1.13beta1 (2016-05-27)

dc42, 2016-06-05:
On reflection, G28 should indeed be allowed, but in simulation mode it needs special treatment:
- don't simulate moving the motors, instead simulate triggering the homing switches
- when existing simulation mode, if the printer wasn't homed when entering it, it isn't homed now.

Kriechi, 2017-11-15:
@dc42 what's the latest implementation here? is this already fixed in 1.19 or 1.20betas?

dc42, 2023-05-04:
Need to test this to see if we can simulate a file that uses G28 at the start when the machine is not already homed.

closed 2023-05-24 by T3P3