Duet3D/RepRapFirmware #49: New Feature - Hobby Servo Support

Hotaman, 2016-06-05:
While there are many new methods to do contactless Z probing now, I still prefer the reliability and simplicity of my gravity based IR probe. I currently get deviations of 0.002 with this probe.

My probe uses a micro 3.3v hobby servo like the ones used on a nano sized heli to hold the probe up when not in use. I currently have to deploy/retract it manually which is a real pain.

I'm looking at adding M280 to the firmware and hijacking the Fan1 code since I don't currently use it on this printer.

WZ9V, 2016-06-05:
I would second this request. It would allow other sensors such as the BLTouch or a mechanically deployed switch.

Hotaman, 2016-06-06:
I'm making some progress on this one and I have added some extra Fan features as a way to get my toes wet.

I added a 'V' parameter to set the On state fan speed when thermostatic mode is set and modified the normal speed set code to allow changing the speed on thermo-switched fans as long as it is not below the new thermo on speed when above trigger temp. The default fan speed is still 100% so these mods have no effect unless you first use M106 Vnnn to set the thermo switch speed below 100%.

Use case:
My printer only has one fan on the hot end which does double duty. I like it to run at a minimum of 40% to keep the hot end cool without too much part cooling or noise, yet still allow the slicer to kick the speed up while printing when more part cooling is needed. The fan must always be on if the hot end is >50C so the min speed should be handled by the thermo switch mode.

The water feels pretty good to my toes so I'm diving into the servo support next. First, I'm going to try to move my servo from end-to-end just using the features of the M106 command. It won't be very positionally accurate, but my use case only needs ~end-to-end movement at this time.

@dc42, Is the arduino servo lib available in the build, just not currently used?

dc42, 2016-06-06:
I have already implemented The S parameter for thermostatic fans, with a minimum of 50% for safety. This feature will be in 1.13 beta 2.

I also have an experimental version of the M42 command, that lets you set a PWM value instead of just on or off. The missing parts are a way of detaching unused heater outputs from the Heat subsystem, so that you are free to use M42 to control them, and a frequency parameter. I think that would be sufficient to control hobby servos. But I still plan to implement M280 or M340.

RRF no longer uses the standard Arduino core.

Hotaman, 2016-06-07:
That's Great! I'll look for the new version :)

I was just using some low hanging fruit on my list to get into the code before doing any real damage. I like your approach to adding proper servo support. I'll continue to explore the code more. Once I get my head all the way around the code I can help out if needed/wanted.

I figured you had moved on from the Arduino core, I didn't see it referenced anywhere.

For anyone interested in using a servo NOW, I have a working solution!

Hook your servo control line to pin 18 on the expansion connector.
Send the following gcodes:

M106 P1 H-1   ; Turn off the default thermo switch on Fan1
M106 P1 S0    ; turn Fan1 off
M106 P1 F50  ; Set 50Hz for the Fan1 frequency, THANK YOU! @dc42 (default is 500Hz)

We can now control the servo by changing the S value of Fan1!

Valid S values: (for my servo, yours may vary)
M106 P1 S14   = ~1ms pulse width
M106 P1 S26   = ~2ms

so S1 = ~83us, the resolution sucks but it is very repeatable so this will work fine to deploy/retract a switch type Z probe to a fixed position. The above values move my servo from min to max throw which is what I need. Your servo may be able to handle up to a 0.5ms - 2.5ms range. You will need to do a bit of experimentation to find the correct values for your setup. When experimenting around the end stops of your servo, make sure to first move it to the middle with S20 before moving to an extreme position to get accurate movement.

I hope this helps someone else to automate their Z probing :)

Hotaman, 2016-06-08:
One final thought, the M106 S param can be a float between 0.0 - 1.0.

So something between 0.05 - 0.1 should give ~1ms - 2ms pulse width. Using floats should give much better positional control when needed.

Enjoy!

mrseanpaul81, 2016-07-30:
I am using a kossel mini that came with the "original" z-probe. That thing was a nightmare so I upgraded to duet board and differential IR. 

What I found is that the differential IR is sensitive to temperature. Due to the fact that I have a heated aluminum bed, the temperature of which can change and I suspect this is the reason why I am getting a changing zprobe height threshold value. This is definitely affecting my prints and driving me nuts. I am sure David did a find job with the differential IR, I just suspect that the heat from the bed is still affecting the probe (even a little bit is enough to mess with things).

I have just ordered a BLTouch as I think this would be better for my needs. I am eager to see proper servo support on RRF and I hope this "work-around" works for me :)

dc42, 2016-07-30:
Hi Jean-Rodney,

The mini IR probe should not be affected at all by temperature, provided
the sensor temperature doesn't exceed about 75C. What sort of variation
are you finding? Have you considered that the heated bed may be flexing
and changing shape due to thermal expansion, or that heat from the hot
end may be affecting whatever mechanism you have used to mount the sensor?

Best regards

David Crocker, Escher Technologies Ltd.
http://www.eschertech.com
Tel. +44 (0)20 8144 3265 or +44 (0)7977 211486

mrseanpaul81, 2016-07-30:
The numbers vary between 3.3 to 3.55 on the web interface (z probe threshold). 
The heated bed flexing may be a very real possibility since I've read somewhere that 3 mm thick aluminum may not be enough. 
The hot end is not as close to IR sensor as I would like and my mounting consists of 3 screws at top securing a custom designed 3d printed part with IR sensor mounted using both holes (imgur pictures: http://m.imgur.com/XdqJ5ql,0J19p3x,8TuQUUY).

dc42, 2016-12-20:
Servo support is implemented in firmware 1.16 and later.

closed 2016-12-20 by dc42