Duet3D/RepRapFirmware #50: 1.13b1: Intermittent extrusion problem

tomasf, 2016-06-06:
I'm running 1.13b1, and sometimes extrusion stops working. The motor doesn't move and no error is printed. It has happened three times now, but I'm not sure how to reproduce it. A reboot fixes it.

    Diagnostics
    Used output buffers: 4 of 32 (28 max)
    Platform Diagnostics:
    Memory usage:
    Program static ram used: 45208
    Dynamic ram used: 40440
    Recycled dynamic ram: 368
    Current stack ram used: 2728
    Maximum stack ram used: 6652
    Never used ram: 5636
    Last reset 17:24:50 ago, cause: software
    Last software reset code & available RAM: 0x0003, 6644
    Spinning module during software reset: GCodes
    Error status: 0
    Bed probe heights: -0.203 -0.092 0.081 -0.139 0.157 -0.109 -0.061 0.063 -0.177 0.017 0.000 0.000 0.000 0.000 0.000 0.000
    Free file entries: 10
    Longest block write time: 124.3ms
    SD card speed: 0.2MHz
    Slowest main loop (seconds): 0.206177; fastest: 0.000000
    Move Diagnostics:
    MaxReps: 6, StepErrors: 0. Underruns: 163
    Heat Diagnostics:
    Heater 0: I-accumulator = 0.0
    Heater 1: I-accumulator = 0.0
    GCodes Diagnostics:
    Move available? no
    Stack pointer: 0 of 5
    macro is idle
    http is ready with "M122"
    telnet is idle
    serial is idle
    aux is idle
    file is idle
    Network Diagnostics:
    Free connections: 14 of 16
    Free transactions: 22 of 24
    Webserver Diagnostics:
    HTTP sessions: 1 of 8
    FTP connections: 2, state 0
    Telnet connections: 1, state 3

tomasf, 2016-06-06:
Okay, it's pretty strange, but here's how I can reproduce it:

1. Start a print
2. Cancel it from Telnet
3. (Extrusion still works at this point)
4. Start a new print
5. Extrusion works for a few seconds and then stops
6. Extrusion is now disabled until reboot

dc42, 2016-12-20:
I suspect what was happening is that cancelling the print turned off the tool heaters, and extrusion stopped when the hot end temperature dropped below the minimum temperature. If that isn't the case and you can reproduce this using firmware 1.17RC1 or later, please re-open this issue.

closed 2016-12-20 by dc42