Duet3D/RepRapFirmware #51: Name

CthulhuLabs, 2016-06-23:
This is just a suggestion.

The name RepRapFirmware is really hard to get useful information about. There is definitely tons of information out there about the project, but search engines seems to think I am searching for general RepRap project information about Firmware. Also many people either do not know about it or get confused when I talk about it because they also think I am talking about firmware for the general RepRap project. I cannot tell you how many times in the last few weeks while setting up RRF on my RADDS board I have had people say, "You mean line Marlin or Repetier?" when asked which firmware I am running.

dc42's fork of the RRF from reprappro has clearly become the default version people should use. I think by changing the name of the firmware it will help to differentiate it better from the original firmware and make the project become more recognizable in the community as a whole.

dc42, 2016-06-24:
Thanks for your suggestion. We considered changing the name during the Duet WiFi project, but we decided against it because the name is so well known. However, we have registered domain name reprapfirmware.org, and we are planning to host all the documentation there. That should make it easier to find.

CthulhuLabs, 2016-06-24:
Yeah that would definitely help, especially if the search engine optimizations are good.

dc42, 2016-10-22:
For additional information about configuring RepRapFirmware, please see https://duet3d.com/wiki/Duet_Wifi_Wiki#Initial_configuration_.28Duet_.26_Duet_WiFi.29. It's aimed at users of the Duet series controller boards,but most of it should be applicable to RepRapFirmware on RADDS too.

closed 2016-10-22 by dc42