Duet3D/RepRapFirmware #52: FeatureRequest - Babystepping

mirage335, 2016-06-27:
"Babystepping", as implemented by Marlin firmware, allows fine adjustment to the machine's position in realtime, during printing, through a control knob. Among other uses, this can enable minor corrections to first-layer alignment by adjusting the Z-axis.

Similar functionality, perhaps in the form of a g-code that could be called by macro from the PanelDue, would bring similar functionality to the Duet platform.

mhackney, 2016-12-21:
I recently built a Prusa i3 MK2 that has this feature and it is VERY useful. It makes tuning the first layer while the skirt is printing. The rotating knob is really important for this feature though. I believe this could be implemented in a touch screen or web UI with a slider bar. The center would be the current Z and sliding left decreases Z and right increases it.

dc42, 2016-12-21:
This feature is already at the top of the wish list and is likely to be implemented in firmware 1.18.

mirage335, 2016-12-21:
Awesome.

dc42, 2017-03-15:
This is implemented in RRF 1.18beta2, PanelDue 1.16beta1 and DuetWebControl 1.15beta2.

closed 2017-03-15 by dc42

mirage335, 2017-03-15:
Awesome!