Duet3D/RepRapFirmware #53: Dual head compatible?

ThorsenRune, 2016-07-14:
Hi, could/should I upgrade my ormerod 1 with dual color extension
FIRMWARE_VERSION:1.09 ELECTRONICS:Duet (+ Extension) DATE:2015-04-21

to this firmware?

dc42, 2016-07-14:
Yes, you can use my 1.14 release on your Ormerod. You will need to define the endstop switch locations with a M574 command in config.g. If your PC runs Windows then you should install the device driver in the /drivers folder. The best place to get support is on the Duet forum at reprap.org.

ThorsenRune, 2016-07-15:
Thank you for your answer but it baffles me. I have a dual head running the mentioned firmware. Why should I install a device driver now if I just wanted to upgrade the firmware?
Why should I define endstop switch location?
Thanks for your time

dc42, 2016-07-15:
The new firmware has many improvements compared to the version you are running and supports a much wider range of 3D printers. One of the improvements requires that the firmware know which end of each axis the endstop switches are located. The default is low end as on the Ormerod 2 and most other 3D printers, so for an Ormerod 1 you need to tell it that you have a high-end Y endstop switch. The device driver is needed when using versions of Windows 10 because the core hardware interface code uses a later version of the Atmel Software Framework than the old one and no longer pretends to be an Arduino Due.

Please ask any further support questions on the forums. This is the place reporting bugs and feature requests, not for general support.

closed 2016-07-15 by dc42

ThorsenRune, 2016-07-15:
Thanks, all clear. Sorry I didn't find the right forum

dc42, 2016-07-15:
The most suitable forum is http://forums.reprap.org/list.php?416.