Duet3D/RepRapFirmware #54: M572 Loosing steps in XY

franklima260, 2016-08-01:
I have tested my firmware config with both M572 activated and deactivated, it seems to cause steps to be lost in the XY when turned on.

Please let me know if I can send any info to help diagnose the problem. I am using this line:

M572 D0 S0.1	; compensate for springiness

dc42, 2016-08-01:
Do you have a reproducible sample print? I honesty can't see any mechanism for this type of problem to occur. The XYZ and extruder movements are computed independently, except that enabling pressure advance may limit extruder acceleration, which could in turn limit XY acceleration so as to stay in sync.

franklima260, 2016-08-02:
I have been able to reproduce the problem consistently with multiple Gcode files. The files works fine and will print all the way through. If I send the M572 command at any point during the print the printer will begin losing steps.

On a side note the firmware also seems to lockup if I use 4 leveling point, works fine with 3 or 5.

Thank you so much for developing such an incredible controller and firmware.

Here is a dump of my config file:

; CoreXY sample config file for dc42 Duet firmware

M111 S0                             ; Debug off
M550 Pworkerbee			        ; Machine name (can be anything you like)
M551 Preprap                        ; Machine password (used for FTP connections)
M540 P0xBE:0xEF:0xDE:0xAD:0xFE:0xED ; MAC Address
;*** Adjust the IP address and gateway in the following 2 lines to suit your network
M552 P0.0.0.0						; IP address (0 = use DHCP)
M554 P192.168.1.1                   ; Gateway
M553 P255.255.255.0                 ; Netmask
M555 P2                             ; Set output to look like Marlin
M575 P1 B57600 S1					; Comms parameters for PanelDue

; Machine configuration
M569 P0 S1							; Drive 0 goes forwards (change to S0 to reverse it)
M569 P1 S0							; Drive 1 goes forwards
M569 P2 S1							; Drive 2 goes forwards
M569 P3 S0							; Drive 3 goes forwards
M569 P4 S1							; Drive 4 goes forwards
; If you use an endstop switch for Z homing, change Z0 to Z1 in the following line, and see also M558 command later in this file
M574 X1 Y2 S1					; set endstop configuration (X and Y and endstops only, at low end, active high)
M574 Z1 S0
M574 E S0
M667 S1								; set CoreXY mode
M92 X81 Y81 Z3200					; Set axis steps/mm
M92 E100						; Set extruder steps/mm
M906 X2200 Y2200 Z2000 E1200            ; Set motor currents (mA)
M201 X4500 Y4500 Z20 E1600            ; Accelerations (mm/s^2)
M203 X32000 Y32000 Z100 E3600       ; Maximum speeds (mm/min)
M566 X3600 Y3600 Z30 E20              ; Maximum jerk speeds mm/minute
;M572 D0 S0.04	; compensate for springiness
M208 X370 Y370 Z400					; set axis maxima (adjust to suit your machine)
M208 X0 Y0 Z-0.15 S1				; set axis minima (adjust to make X=0 and Y=0 the edges of the bed)
G21                                 ; Work in millimetres
G90                                 ; Send absolute coordinates...
M83                                 ; relative extruder moves

; Z probe
M558 P4 X0 Y0 Z1                    ; Analog Z probe, also used for homing the Z axis
G31 Z-0.15 			                    ; Set the probe height and threshold (put your own values here)
; The following M557 commands are not needed if you are using a bed.g file to perform bed compensation
;*** Adjust the XY coordinates in the following M557 commands to suit your build and the position of the Z probe
M557 P0 X0 Y370                        ; Four... 
M557 P1 X355 Y370                    ; ...probe points...
M557 P2 X355 Y0                   ; ...for bed...
M557 P3 X185 Y185                     ; ...levelling
M557 P4 X10 Y10 

; Thermistors and heaters
;*** If you have a Duet board with 1K thermistor series resistors, change R4700 to R1000 to the following M305 commands
; You can also use S and B parameters to define the parameters of the thermistors you are using
M305 P0 T100000 B4334 R4700 H0 L0					; Put your own H and/or L values here to set the bed thermistor ADC correction
M305 P1 T1000000 B4926 R4700 H30 L0						; Put your own H and/or L values here to set the first nozzle thermistor ADC correction
M305 P2 R4700 H0 L0					; Put your own H and/or L values here to set the second nozzle thermistor ADC correction
M304 P125 I5 D500 T2	; T is count per deg c feedforward 
M301 H1 P30 I3 D250 T1	; PID settings for extruder 0
M301 H2 P10 I0.10 D100 T0.50 S1.0	; PID settings for extruder 1
M570 S120							; Increase to allow extra heating time if needed
M143 S300	; Set maximum hot end temperature

; Tool definition
M563 P0 D0 H1                       ; Define tool 0
G10 P0 S0 R0                        ; Set tool 0 operating and standby temperatures
;*** If you have a dual-nozzle build, un-comment the following 3 lines
;M563 P1 D1 H2                      ; Define tool 1
;G10 P1 S0 R0                       ; Set tool 1 operating and standby temperatures

;*** If you are using axis compensation, put the figures in the following command
M556 S78 X0 Y0 Z0                   ; Axis compensation here
T0									; select first hot end

M500 ;Store setting in EEPROM

dc42, 2016-08-02:
Please provide a small gcode file that demonstrates the problem, and I will try to reproduce it. Preferably a regular shape (e.g. cube, cylinder etc.) so that lost steps can be seen clearly. Also provide an image of a print of that demonstrating the problem, where you had no pressure advance configured initially but sent M572 D0 S0.1 after printing a number of layers.

ccavanaugh, 2016-08-06:
I'm seeing this also on a Delta.  Works fine with Firmware 1.13 but not working with 1.14 and the 1.15 beta. Disabling pressure advance corrects it on 1.14.  

The shift is only in the x&y, z appears to be okay.

Examples attached.  Not exactly simple parts, but demonstrates the issues.

The shift starts on the 5th layer which is infill, the first 4 solid layers are fine.

[SouthBend-BedProtector.txt](https://github.com/dc42/RepRapFirmware/files/404946/SouthBend-BedProtector.txt)

[image]

[image]

dc42, 2016-08-06:
Can you both confirm that you are running on a Duet or Duet WiFi and using the internal drivers? The step pulse generation timing changed in 1.14 and this could affect operation when you are using external drivers.

ccavanaugh, 2016-08-06:
Duet 0.8.5 with onboard drivers

franklima260, 2016-08-06:
Duet 0.8.5 with onboard drivers for me too

franklima260, 2016-10-01:
I can confirm this issue is still present with the Duet Wifi.

dc42, 2016-10-01:
I just printed file SouthBend-BedProtector.txt with pressure advance set to 0.1 and it printed perfectly. The only changes I made to the file were to the temperatures to suit my PLA.

Please can you both do the following tests:

1. Try a print with M572 set to zero but with extruder jerk set to 6 instead of the usual 600 or whatever you are using. This will replicate one of the side effects of having pressure advance set to 0.1.

2. Run the following commands and then try a print with M572 set to 0.1 or whatever usually gives the problem:

M569 P0 T2
M569 P1 T2
M569 P2 T2

I have another report that may be related (see http://forums.reprap.org/read.php?416,704968) and I'll investigate that one next.

dc42, 2016-10-01:
No need for more tests, I can reproduce this problem by setting pressure advance to 0.005. I can hear it skipping twice per layer while printing the hexagonal infill.

dc42, 2016-10-01:
I have fixed this issue in the forthcoming 1.15e release of RepRapFirmware.

ccavanaugh, 2016-10-02:
I've tested with the 1.15e release and it is working well again.  Thanks!

Extruder jerk was/is set at 1200 mm/minute.

closed 2016-10-08 by dc42