Duet3D/RepRapFirmware #55: Triangle interpolation: point outside all triangles!

franklima260, 2016-08-09:
The sheer number of these messages renders the interface unusable. I understand the the error message is harmless but it is really bothersome.

[image]

Is thier any way to disable this error message?

dc42, 2016-08-10:
You should only see this message during homing, if ever. Is that the case? If you are seeing it more often then there may be something wrong with your choice of probe points. The first four should be in an anticlockwise square at the corners of the bed starting near (0,0), and the last one in the centre.

You can set a timeout for messages to disappear on the |Settings page of the web interface.

franklima260, 2016-08-11:
Here are my points:

M557 P0 X50 Y320                        ; first point
M557 P1 X320 Y320                    ; ...probe points...
M557 P2 X320 Y50                   ; ...for bed...
M557 P3 X185 Y185
M557 P4 X50 Y50                     ; ...levelling

Why would the order of the points affect the math?

The screen get covered in the error message more or less continuously during the print.
The problem with making the notification timeout slower is that most notifications are useful and I don't want to miss them.

[image]

dc42, 2016-08-11:
So your points are NOT in the required order (see my previous reply), which explains the problem.

I'll add a firmware wishlist to reject points that are in the wrong order, or to re-order them so that they are.

closed 2016-08-11 by dc42