Duet3D/RepRapFirmware #56: Ethernet connection issues

ShodaPublicity, 2016-08-10:
After updating to v1.14 I frequently get disconnects. Whenever I restart the printer I'll also need to reboot my router (hooked up as a bridge) to be able to access the web interface. Whenever my printer is done printing and if I leave it alone for a while it will also disconnect from the router.

Here are my network setttings:

M552 P192.168.1.10       ; IP address (0 = use DHCP)
M554 P192.168.1.195     ; Gateway
M553 P255.255.255.0     ; Netmask

closed 2016-10-02 by ShodaPublicity

dc42, 2016-10-02:
ShodaPublicity, I see you closed this issue. Does that mean you solved the problem? What was the fix?

ShodaPublicity, 2016-12-19:
For some reason using a static IP address that I had setup with my router was causing issues. After changing the Duet configuration over to using DHCP everything was working fine again. Sorry for the late reply.

dc42, 2016-12-20:
I'm glad you solved it. Perhaps there was an IP address conflict with something else on the network?