Duet3D/RepRapFirmware #57: Is Firmware 1.15 compatible to 3D Artists Alligator Board ?

Jet-A1, 2016-09-11:
Hello,
Marco Antonini from 3D Artists did a port for the Alligator Board with you Reprap Firmware 1.09h-dc42 a long Time ago. My Question is your current Version of Reprap Firmware compatible to the Alligator Board or can you make an compatible Version ?
 
I like the Firmware, because i only need to configure the Printer over the SD Card, not flashing the Board for changes in the Firmware everytime :-)

Greetings

jamesarm97, 2016-09-11:
I have one of these boards that hasn't been used since I bought it because there was no real advantage to using it since there was no firmware at the time that made use of the Ethernet port. I would love to see reprapfirmware run on it. I know some changes lately were to make it easier to adapt to different boards.

dc42, 2016-09-12:
Since version 1.09 the build system for RRF has been completely changed, largely to make it easier to run RRF on other Atmel-based hardware such as Duet WiFi and Arduino Due/RADDS. So adapting it for Alligator should not be difficult, especially as it has been done before. But someone with an Alligator would have to do it.

Jet-A1, 2016-09-12:
Thanks a lot for your Reply,
i own such a Board, but i dont have the Experience to make a Port.
Greetings from Germany

closed 2016-09-12 by Jet-A1