Duet3D/RepRapFirmware #58: Bed heater goes into fault mode despite being tuned

Djhg2000, 2016-09-12:
I have a Kossel with a very weak bed heater. When I'm trying to print I get the following error string:

> Error: heating fault on heater 0, temperature rising much more slowly than the expected 0.0°C/sec

I suspect 0.0°C/sec is a truncated value and the measured value gets truncated to exactly 0.0°C/sec and triggering the heating fault. I can clear the error with `M562 P0` but it keeps triggering as it heats up which gets very tedious and so far I have been unable to reach the target bed temperature of 100°C for my particular blend of ABS.

Is there any way to disable fault checking for a specific heater? If not, may I suggest adding an extra parameter to `M562` for doing so?

jamesarm97, 2016-09-12:
Mine heated up slower and I also got the alarm but was able to adjust the alarm time by issuing an M307 H1 D45 (D45 changed the dead time). You mentioned PID so I am not sure if this will work for you.

Djhg2000, 2016-09-12:
Thank you, D45 was too short but `M307 H0 D349` seems to have done the trick for now (D350 and above threw a "bad model parameters" error).

closed 2016-12-20 by dc42