Duet3D/RepRapFirmware #59: M500 not saving current setting to EEPROM

franklima260, 2016-09-14:
I just switched to the DuetWIFI and am having some major trouble getting the printer to pay nice with my settings. This is to say these settings worked fine on the Duet 0.8.5

At the end of my config.g file I write the setting to EEPROM using M500. If I have my system read back the setting they appear correct, using M503. However if I send to M501 command the setting are changed, the most obvious change is Z prob is wrong but a number of settings are not correct, but when I run M503 it reads back all the correct values. The correct setting seem to load when I send M999.

dc42, 2016-09-14:
The way M500 is implemented does not work on the Duet WiFi and that whole area is being rewritten, probably in time for the next release.

franklima260, 2016-09-15:
Would it be possible to update the reprap g code wiki page to reflex this or/and to the know issues page for the wifi duet?

https://duet3d.com/wiki/Pre-production_hardware_limitations_and_known_firmware_issues
http://reprap.org/wiki/G-code#M500:_Store_parameters_in_EEPROM

dc42, 2016-09-18:
I have updated https://duet3d.com/wiki/Pre-production_hardware_limitations_and_known_firmware_issues#Known_firmware_issues.

dc42, 2016-12-20:
This is fixed in firmware 1.17RC1.

closed 2016-12-20 by dc42