Duet3D/RepRapFirmware #60: FeatureRequest - Dual Independent X-Carriages

kearneylackas, 2016-09-23:
I've seen several printers lately running dual X-carriages (mainly https://www.bcn3dtechnologies.com/en/catalog/bcn3d-sigma/). Advantages include no drooping from the inactive extruder and no dragging the inactive extruder on the print due to slight height differences. We would need additional drive electronics but the expansion header could easily be leveraged.

shadow651, 2016-09-23:
Interesting concept. I'd add that if you have a dueX4 expansion board and
you don't plan on using all of those steeper drivers for extrusion you
could use one of those.

We'd need to know the offsets between the rails the heads ride on and the
park location for the inactive head.

kearneylackas, 2016-09-23:
Well... I am designing a new printer right now actually and want to use Duet Wifi. Dual X-axis is implemented in the Marlin firmware right now but if it is on the roadmap I would prefer to use this board. I don't think the dueX4 is compatible with Duet Wifi but it is easy enough to dump the right pins into an external stepper driver. It would only take one additional driver I believe.

T3P3, 2016-09-23:
This is a feature request that will be implemented, for more information
see:

https://www.duet3d.com/forum/thread.php?id=8

The planned Duex5 will provide up to 5 more stepper channels so will
support this and more!

kearneylackas, 2016-09-23:
Awesome!

dc42, 2016-10-22:
This feature had been implemented in the latest bets (1.16 beta 4) and is being tested.

dc42, 2016-10-22:
See https://duet3d.com/wiki/Configuring_multiple_independent_X-carriages_on_a_Cartesian_printer for how it works. There is a thread about it here https://www.duet3d.com/forum/thread.php?id=340.

closed 2016-12-20 by dc42