Duet3D/RepRapFirmware #63: Feature request: fan mininimal value and fan blip

Alex9779, 2016-10-12:
This is something Marlin has which is pretty nice.

First I want to define a minimal value a fan has to have to run reliable without having to take care of that in the slicer.
So for example my fan runs at S30 reliable which is about 11% I wanna be able to tell that the firmware so even if I send M106 S15 the fan runs at least at S30. So everything from S1 to S30 will be S30.

Second my fan does not run if I just set it to 20%, it needs a "blip". In Marlin I can set that the fan should run at 100% for some milliseconds when it is powered the first time.
S3D has such an option to do it with Gcode but I don't like that option because it sets the fan to 100% and then waits for 500ms which stops all movements and that results in blobs and melts on the print...

dc42, 2016-10-12:
I've added this to the work list. it should be straightforward to do, but I'll need to add new parameters to M106 to define the minimum speed and the blip time.

Alex9779, 2016-10-12:
Pretty cool, thanks!

> I'll need to add new parameters to M106 to define the minimum speed and the blip time

Since you doing the whole config with gcodes I assumed that :D

Alex9779, 2016-10-19:
Nice, it is in 1.16. Closing here, will test soon...

closed 2016-10-19 by Alex9779