Duet3D/RepRapFirmware #65: FeatureRequest - M143 with multiple source and values

irpyc, 2016-11-13:
Hey,
First of all, really great job! I just get my Duet Wifi for some weeks, no way that I come back on Rumba...

As I see that you're right now working on temperature stuffs, I would be pleased to get a M143 with more functions.
Ok, both of my Hot End are the same, and I guess it's like this in most of the cases, but what I would like is to be able to set a maximum temperature to any temperature sensor connected.

My printer is in a chamber, and for instance, if the temperature at this sensor exceed 80°C, that will just mean something goes wrong.
I guess setting such limit on the Bed could make sense as well.
I was also thinking about putting a second thermistor on the heat sink of the Hot End, this could both prevent fan and primary thermistor failure, but for this second point the firmware seems to have already a lot of nice protections, so maybe this is just paranoia...

To go further, it would be really nice also if any of theses emergency stop (actioned by the firmware itself) could lead to a warning, the most adaptive could be a "emergency_stop.g" script, then M42, latching relay and buzzer would make it. (It's seems that the PanelDue have a buzzer, but I don't have £60 for this, especially since you can get an Android tablet second hand three time cheaper).

I can hardly program but let me know for testing.

I was also looking for a M460 but I just realized that M106 can do it. I'll ask for a wiki account to put a reference there.

dc42, 2016-11-13:
M143 with per-heater limits will be in the next 1.17 dev build.

There is a plan to support a fault.g file that would be run when over temperature or any other heating fault occurs. I'm not sure whether this will make it into version 1.17.

irpyc, 2016-11-14:
Sound's perfect! Great job
Thanks for the infos.

closed 2016-11-14 by irpyc