Duet3D/RepRapFirmware #67: Release 1.16 Runs for for hour, then all axes start to Chader

gdwinslow, 2016-11-17:
Started a print.  all Axes where working OK.  it was a 50mm X 40mm Cylinder.  At Z=20mm, all the axes motors started to cheater.  At z=40mm the Extruder cheaters so bad it strips the filament.  Printer finishes print, and shuts down normally.  I ran a Firmware Diagnostics and included it after the configuration.

**Current Configuration:**

M111 S2                                          ; Debug on
M550 PTricolour Mendel WIFI
M551 Pgdww1620
M540 P0xBE:0xEF:0xDE:0xAD:0xFE:0xEA              ; Set the MAC address
M552 S1				                             ; Enable WiFi

M555 P2                                          ; Set output to look like Marlin
M575 P1 B57600 S1                                ; Comms parameters for PanelDue
M906 X800 Y1000 Z800 E800:800                    ; Motor currents (mA)

M305 P0 R4700 T10000 B3988                       ; set bed thermistor parameters
M307 H0 A117.7 C395.8 D1.8 B0                    ; set Heater 0 PID
M307 H1 A662.1 C75.5 D5.9 B0                     ; set Heater 1 PID

M574 X0 S1                                       ; Set endstop configuration
M574 Y2 S0                                       ; Set endstop configuration
M574 Z0 S1                                       ; Set endstop configuration

M569 P0 S0                                       ; Set X Axis Direction
M569 P1 S1                                       ; Set Y Axis Direction
M569 P2 S0                                       ; Set Z Axis Direction
M569 P3 S0                                       ; Set E0 Axis Direction
M569 P4 S0                                       ; Set E1 Axis Direction

M558 P1 X1 Y0 Z1                                 ; Set Z probe type
M92 X100.25 Y100.25 Z4001.50 E420.00:420.00      ; Set axis steps per unit
M570 S180                                        ; Set heater timeout
M667 S0                                          ; set Cartesian mode

M350 X0 Y0 Z0 E0:0 I0                            ; Set microstepping mode

M563 P0 D0 H1                                    ; Create tool 0 = Color 1
M563 P1 D1 H1                                    ; Create tool 1 = Color 2
M563 P2 D2 H1                                    ; Create tool 2 = Color 3

G10 P0 X00.0 Y0.0 Z0.0 S0 R0                     ; Set tool 0 operating and standby temperatures
G10 P1 X00.0 Y0.0 Z0.0 S0 R0                     ; Set tool 1 operating and standby temperatures
G10 P2 X00.0 Y0.0 Z0.0 S0 R0                     ; Set tool 2 operating and standby temperatures

M106 P2 T45 H1                                   ; Fan control, Thermostatic on Heater H1
M106 P0 F75 B1                                   ; Fan control, Cooling fan

M572 D0 S0.05                                    ; Set extruder 0 pressure advance
M572 D1 S0.05                                    ; Set extruder 1 pressure advance

M201 X800 Y800 Z15 E800:800                      ; acceleration tweaks for improved finish
M203 X15000 Y15000 Z150 E3600:3600               ; speed tweaks for improved finish

M208 X200 Y200 Z120                              ; set axis max travel
M208 X0.0 Y0.0 Z0.0 S1                           ; set axis minimum (adjust to make X=0 the edge of the bed)

G21                                              ; Work in millimetres
G90                                              ; Send absolute coordinates...
M83                                              ; relative extruder moves

G31 P500 Z0.35                                   ; Set the probe height and threshold PLA = 0.65, ABS = 0.75
T2
M500                                             ;Store setting in EEPROM

**Firmware Diagnostics:**
7:46:59 AMM122
=== Diagnostics ===
Used output buffers: 1 of 32 (5 max)
=== Platform ===
Memory usage:
Program static ram used: 19700
Dynamic ram used: 85852
Recycled dynamic ram: 944
Current stack ram used: 2792
Maximum stack ram used: 7888
Never used ram: 16688
Last reset 00:07:48 ago, cause: power up
Error status: 0
Bed probe heights: 0.150 0.221 -0.114 0.023 0.017 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000
Free file entries: 9
SD card 0 detected, interface speed: 20.0MBytes/sec
SD card longest block write time: 0.0ms
MCU temperature: min 29.9, current 36.9, max 39.6
Supply voltage: min 14.6, current 15.2, max 15.4, under voltage events: 0, over voltage events: 0
Driver 0: stalled
Driver 1: stalled
Driver 2: stalled
Driver 3: standstill
Driver 4: stalled open-load-A
Current date and time: 2016-11-15 07:46:35
Slowest main loop (seconds): 1.059487; fastest: 0.000038
=== Move ===
MaxReps: 4, StepErrors: 0, MaxWait: 0ms, Underruns: 0, 1
=== Heat ===
Bed heater = 0, chamber heater = -1
Heater 0 is on, I-accum = 0.3
Heater 1 is on, I-accum = 0.2
=== GCodes ===
Move available? yes
Stack pointer: 0 of 5
macro is idle
http is ready with "M122"
telnet is idle
serial is idle
aux is idle
file is idle
=== Network ===
WiFiServer is running
SPI underruns 0, overruns 0
=== Webserver ===
HTTP sessions: 1 of 8

dc42, 2016-11-17:
I don't think this has anything to do with the firmware. It may be a power problem. Check that the screws in the VIN terminal blocks are tight.

Also make sure that the Duet is not getting hot. It won't generate much heat itself at the motor currents you are using, but it could be picking up heat from elsewhere e.g. if you have put it beneath an un-insulated heated bed. You can check that the stepper drivers are not overeheating when the printer is running by sending M122 and looking at the Driver status.

The M350 line in your config.g file is wrong. It should be:

 M350 X16 Y16 Z16 E16:16 I1 ; Set microstepping mode

The firmware is almost certainly ignoring that line, so I doubt that it is related to this issue.

gdwinslow, 2016-11-17:
Turning on microstepping mode, makes the problem worse,  axes cheater all the time, Evan from the start. The power supply is under the table top from the printer.  The printer board is in a Box with know lid, mounted on the side of the printer, open side facing away from the printer.

I trued this first:
M350 X16 Y16 Z16 E16:16 I1 ; Set microstepping mode
but as i said all axes cheater from the start.  Setting I1 is loader then I0 in the M350 command

it is possible the VIN terminal blocks is defective like the Heat bead terminal block was.  I had to replace the Heat bead terminal block, as it was jamming instead of clamping down on the Wires.

Note - This power supply was driving my old Duet board before i put the DuetWiFi board on the printer.

Dos the DuetWiFi need a heaver supply?  If so what is the recommended power supply spec?
My current power supply spec is: 15V 23.5A

gdwinslow, 2016-11-18:
OK, doubled the wires going from my power supply to the board and set the microstepping mode as suggested. I also set the motor currents all the same.  it started out sounding like its a Tran going down the Tran tracks, Clickety clack.  At the start.  after cupple of layers it started to sound normal.  hear is the M122 command dump i took after the first layer:

8:33:36 PMM122
=== Diagnostics ===
Used output buffers: 1 of 32 (7 max)
=== Platform ===
Memory usage:
Program static ram used: 19700
Dynamic ram used: 85964
Recycled dynamic ram: 832
Current stack ram used: 2792
Maximum stack ram used: 7952
Never used ram: 16624
Last reset 00:05:47 ago, cause: power up
Error status: 0
Bed probe heights: 0.041 0.219 -0.079 -0.058 0.017 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000
Free file entries: 9
SD card 0 detected, interface speed: 20.0MBytes/sec
SD card longest block write time: 0.0ms
MCU temperature: min 36.0, current 39.1, max 42.5
Supply voltage: min 14.8, current 15.3, max 15.4, under voltage events: 0, over voltage events: 0
Driver 0: stalled
Driver 1: stalled
Driver 2: stalled
Driver 3: standstill
Driver 4: stalled open-load-A
Current date and time: 2016-11-17 20:33:20
Slowest main loop (seconds): 1.084417; fastest: 0.000064
=== Move ===
MaxReps: 4, StepErrors: 0, MaxWait: 0ms, Underruns: 0, 0
=== Heat ===
Bed heater = 0, chamber heater = -1
Heater 0 is on, I-accum = 0.3
Heater 1 is on, I-accum = 0.2
=== GCodes ===
Move available? no
Stack pointer: 0 of 5
macro is idle
http is ready with "M122"
telnet is idle
serial is idle
aux is idle
file is idle
=== Network ===
WiFiServer is running
SPI underruns 0, overruns 0
=== Webserver ===
HTTP sessions: 1 of 8

gdwinslow, 2016-11-18:
OK, the doubled up on my 12 gauge wire seems to be doing the trick.  this is the Best and only good print i have gotten out of this board so far.  This is the M122 Dump i did at 48mm of a 50 mm cylinder:

11:23:54 PMM122
=== Diagnostics ===
Used output buffers: 1 of 32 (11 max)
=== Platform ===
Memory usage:
Program static ram used: 19700
Dynamic ram used: 85964
Recycled dynamic ram: 832
Current stack ram used: 2792
Maximum stack ram used: 7952
Never used ram: 16624
Last reset 02:56:05 ago, cause: power up
Error status: 0
Bed probe heights: 0.041 0.219 -0.079 -0.058 0.017 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000 0.000
Free file entries: 9
SD card 0 detected, interface speed: 20.0MBytes/sec
SD card longest block write time: 0.0ms
MCU temperature: min 37.2, current 39.8, max 43.7
Supply voltage: min 14.7, current 15.2, max 15.5, under voltage events: 0, over voltage events: 0
Driver 0: stalled
Driver 1: stalled
Driver 2: stalled open-load-B
Driver 3: standstill
Driver 4: stalled
Current date and time: 2016-11-17 23:16:28
Slowest main loop (seconds): 0.009033; fastest: 0.000000
=== Move ===
MaxReps: 5, StepErrors: 0, MaxWait: 1ms, Underruns: 0, 0
=== Heat ===
Bed heater = 0, chamber heater = -1
Heater 0 is on, I-accum = 0.2
Heater 1 is on, I-accum = 0.2
=== GCodes ===
Move available? yes
Stack pointer: 0 of 5
macro is idle
http is ready with "M122"
telnet is idle
serial is idle
aux is idle
file is idle
=== Network ===
WiFiServer is running
SPI underruns 0, overruns 0
=== Webserver ===
HTTP sessions: 1 of 8
11:23:49 PMMessage Log cleared!

gdwinslow, 2016-11-19:
Ok, ran some speed change test.  it appears to Like to print objects sliced at 24mm/s.  Anything more or lease and it cheaters on all axes.

dc42, 2016-11-19:
Please raise this and other general support questions on the forum at duet3d.com or reprap.org. This github issues section is for confirmed firmware bugs only.

closed 2016-11-19 by gdwinslow