Duet3D/RepRapFirmware #70: Addition of M571 options

timmit99, 2016-12-01:
Would it be possible to add a "P" parameter to the m571 G-code? This could be used to specify which pin will be toggled on extrude. If it is not a PWM pin it could just turn the pin on/off. I am currently using both fans on the PWM pins for other uses and would like to implement the trigger on extrude option but cannot give up Fan0's slot.

On a related note, where should i post to ask questions about the web interface? Specifically about adding a second fan slider for the PWM pins and adding some other custom G-code commands to the sections.

dc42, 2016-12-01:
1. I'll add that to the work list.

2. Either in the Duet section of the reprap.org forum, or at chrishamm's DuetWebControl GitHub repo.

dc42, 2016-12-20:
This is implemented in firmware 1.17RC1,

closed 2016-12-20 by dc42