Duet3D/RepRapFirmware #71: Firmware 1.17 dev6 , freezes after manuall Z-movement during printing

Pumlux, 2016-12-06:
I noticed that the printing freezes if I make a manuall z-movement during printing.

In former version on the firmware ( I think it was 1.12a on the 0.8.5) I used to correct the z-height
manually during the first layer printing to find a z-height with good layer adhesion to the bed.

Trying the same on the Wifi with 1.17 dev6 results in a stop of the print.

dc42, 2016-12-06:
I did a quick test and I can't reproduce this. Please provide more detail.

Pumlux, 2016-12-06:
You are right.
I use selfdefined macros to move z-axis  in +- 0.01 mm steps.
This includes also M300 command.

------------- Macro - 0.01-------
G91
G1 Z-0.01 F900

G90
M300 S200 P500
M300 S800 P150
M300 S1500 P100
-----------------------------------
------------- Macro + 0.01-------
G91
G1 Z0.01 F900

G90
M300 S2000 P150
M300 S400 P450
M300 S150 P900
----------------------------------

dc42, 2016-12-06:
I have reproduced the issue. Looking to fix it in the next 1.17 dev build.

dc42, 2016-12-20:
Please re-test with firmware 1.17RC1 or later.

Pumlux, 2016-12-24:
I tested with firmware 1.17rc2 and can confirm this issue is solved here.
No longer freezing of the printing process.
Thanks for your hard work.

And have a nice X-Mas.

closed 2016-12-24 by Pumlux