Duet3D/RepRapFirmware #72: Cura 2.3.x slicer identification

ccavanaugh, 2016-12-17:
Cura 2.3.x is using  ";Generated with Cura_SteamEngine master" to identify the slicer.  

PrintMonitor::GetFileInfo needs to be updated to identify the slicer correctly.

I'm assuming "master" can be ignored...

dc42, 2016-12-20:
If Cura wants to report itself as Cura_SteamEngine_master, that's how RRF is going to report it. I suggest you talk to the developers of Cura if you want to change that.

ccavanaugh, 2016-12-21:
I'm not so much worried about how it displays, but it does not display it at all.  The firmware does not identify it as a slicer and leaves the field blank.

dc42, 2016-12-21:
Can you provide a sample file?

ccavanaugh, 2016-12-21:
Sample file attached.

Also, Cura no longer reports filament usage in the file which is frustrating.  The firmware code to estimate filament usage is not working either.  The pull request I submitted works for identifying the Cura, but is not necessarily the best approach.. just simple.

Thanks!

[W_Tie Down.txt](https://github.com/dc42/RepRapFirmware/files/666033/W_Tie.Down.txt)

dc42, 2016-12-21:
Fixed in 1.17RC3. I can't do anything about the filament usage, that's down to the Cura devs to add the filament usage comment in the file.

closed 2016-12-21 by dc42