Duet3D/RepRapFirmware #76: Feature Request: Stateful spool tracking

kag86, 2017-01-14:
First off, great work on this firmware fork!  Onto my request/thought.

I was just changing a spool today and thought it might be nice if the firmware and web interface had some sort of filament spool states associated with it.  These would basically track extruder movement with something as simple as counters that track extruder movements since the last spool change event was issued to the web interface.  Ideally updates to this state would be periodically written to the sdcard to keep the estimate fairly accurate.

This feature might be useful for estimates of filament remaining, inventory tracking, and perhaps for optionally auto-pausing a print near the end of spool for those of us without filament detectors.  I've haven't really taken much of a look at the source for this firmware, but I could probably hack up a rough start for what I'm suggesting if you'd like.  I haven't written firmware in about 5 years though, and I'm not terribly familiar with this arduino platform, web programming in general or the build environment.

Regards,
Kyle

shoreliner, 2018-09-01:
This should be fairly simple to implement -- we've been looking at providing something like this from the M3D side. I'll see what I can do once our current launch is over!

dc42, 2018-09-02:
Please consult with me and chrishamm first so that we can agree a specification. See https://duet3d.dozuki.com/Wiki/Filaments for what RRF already supports in this area.

digi2life, 2018-09-02:
I'm all for this feature addition. Markforged does this exactly by having the user enter spool weight when loading filament. The print is auto paused when the spool runs low. You need to know empty spool weight and make sure the remaining amount is stored before power cycle. It works very well.

Nightreaver, 2019-04-24:
If I may add... after coming from Octoprint.. I was thinking about that at first - https://plugins.octoprint.org/plugins/filamentmanager/

I was a bit sas when it turned out "just" to be for filament specific gcode

T3P3, 2020-10-20:
This is a candidate for a plugin - please discuss further on the forum.

closed 2020-10-20 by T3P3