Duet3D/RepRapFirmware #78: Feature Request: Add support for DUE3DOM MINI

dvjcodec, 2017-01-29:
Is possible to add DUE3DOM MINI board to RepRapFirmware?
https://github.com/dvjcodec/Due3dom-Mini

dc42, 2017-01-29:
It should be possible to adapt the RADDS configuration to work with that board. Sadly the Arduino Due doesn't provide an Ethernet port, so you would lose one of the major features of RepRapFirmware.

dvjcodec, 2017-01-30:
I wonder to make extension with Ethernet in future, but now havn't :( 
Can you adapt configuration for this MINI board? What can I help with this issue? Do you need MINI board?

dc42, 2017-02-05:
I don't have time to port RRF to new boards, but I'll happily provide advice to anyone else who wants to do the port. Changes will be needed to at least 2 source files, but possibly no more than that.

dc42, 2017-03-15:
btw if you use the Wiz850io instead of the Wiz820io then the port will be easier, because the new Duet Ethernet uses the same chip (W5500) as the Wiz850io.

closed 2017-03-25 by dc42