Duet3D/RepRapFirmware #79: RepRapFirmware compile error

ymilord, 2017-02-22:
I made a fresh dev environment that I created last night with eclipse. 
When attempting to build the firmware (fresh git pull from this repo with no changes and after building CoreNG first with no errors) I get this-

../src/Movement/Move.cpp: In member function 'void Move::DoDeltaCalibration(size_t, StringRef&)':
../src/Movement/Move.cpp:1066:2: error: 'float_t' was not declared in this scope
../src/Movement/Move.cpp:1066:10: error: expected ';' before 'initialSumOfSquares'
../src/Movement/Move.cpp:1078:3: error: 'initialSumOfSquares' was not declared in this scope
../src/Movement/Move.cpp:1196:32: error: 'initialSumOfSquares' was not declared in this scope
make: *** [src/Movement/Move.o] Error 1

Any suggestions?

dc42, 2017-02-22:
In my build environment, 'float_t' is defined in the math.h library that is part of the gcc installation. However, the use of 'float_t' was unintentional, and I will change it to 'float' in the next commit.

ymilord, 2017-02-24:
Ah okay thanks.

I just did a pull and now the current error is-

Building file: ../CoreNG/variants/duetNG/exceptions.c
../CoreNG/variants/duetNG/exceptions.c:58:21: fatal error: wdt/wdt.h: No such file or directory
compilation terminated.
make: *** [CoreNG/variants/duetNG/exceptions.o] Error 1

dc42, 2017-02-24:
exceptions.c is compiled in the CoreNG project, not the RepRapFirmware project.

ymilord, 2017-02-24:
Yeah, I just realized that. Corrected that and now I am getting-

In file included from ../Fan.h:11,
                 from ../Fan.cpp:8:
../RepRapFirmware.h:29: fatal error: ecv.h: No such file or directory
compilation terminated.
make: *** [Fan.cpp.o] Error 1

dc42, 2017-03-15:
ecv.h is in cores/Arduino in the CoreNG project. The RepRapFirmware project includes that folder in its include path. It sounds to me that you have not followed the instructions for setting up your workspace. You need ONE workspace with TWO projects (CoreNG and RepRapFirmware) in it.

closed 2017-03-15 by dc42

SijieLiang, 2018-11-15:
> Yeah, I just realized that. Corrected that and now I am getting-
> 
> In file included from ../Fan.h:11,
> from ../Fan.cpp:8:
> ../RepRapFirmware.h:29: fatal error: ecv.h: No such file or directory
> compilation terminated.
> make: *** [Fan.cpp.o] Error 1

I just had the same problem to compile the CoreNG project, how did you solve it?

SijieLiang, 2018-11-15:
> Ah okay thanks.
> 
> I just did a pull and now the current error is-
> 
> Building file: ../CoreNG/variants/duetNG/exceptions.c
> ../CoreNG/variants/duetNG/exceptions.c:58:21: fatal error: wdt/wdt.h: No such file or directory
> compilation terminated.
> make: *** [CoreNG/variants/duetNG/exceptions.o] Error 1

I just had the same problem to compile the CoreNG project, how did you solve it?