Duet3D/RepRapFirmware #80: Feature request: Select extruder/tool for M571 (Set output on extrude)

timmit99, 2017-02-27:
I was hoping there could be a parameter (D) that would allow you to select which extruder/tool you want to trigger the output on extrude. That way you can have the output go high only when a selected tool is operational.

dc42, 2017-02-28:
Can you tell me that application that you have in mind?

You could use M571 commands in the tool change files to enable/disable output depending on which tool is selected, but I appreciate that you might want something more secure if you are controlling a laser.

timmit99, 2017-02-28:
yes, i have a combo laser cutter/mill and i sometimes do 2 operations in a single file. My first "extruder" (tool) is the laser cutter and the second "extruder" (tool) is a spindle. I will etch the plate and then mill the outer edges to shape, being able to have the laser on ONLY when i'm using the laser section would be nice, as of now i'm manually going through the code and triggering it on and off with a python script. If i could assign the "Trigger on extruder" just for the laser that would be ideal.

T3P3, 2023-05-19:
this would also be useful for an academic project.