Duet3D/RepRapFirmware #84: DuetWiFiFirmware-1.18beta2 has bug with X and Y homing on CoreXY

yonkiman, 2017-03-16:
(Apologies for originally posting this in the Duet Wifi General forum - I just haven't been thinking very clearly today.)

My CoreXY printer was very stable with DuetWiFiFirmware-1.18beta1, but after flashing DuetWiFiFirmware-1.18beta2, homing broke.   When doing an X home or a Y home, only one of the two motors would turn.  Flashed back to DuetWiFiFirmware-1.18beta1 and everything's fine.

```
; X axis homing file

G91			; relative mode
G1 Z4.0 F6000		; lower bed to avoid dragging nozzle over the bed
G1 X-240 F2000 S1	; move up to 240mm in the -X direction, stopping if the homing switch is triggered
G1 X2 F2000		; move slowly  2mm in the +X direction
G1 X-10 F 360 S1	; move slowly 10mm in the -X direction, stopping at the homing switch

G90			; absolute mode
G1 X0 F2000		; Move head to 0 (corner of bed)
G91			; relative mode
G1 Z-4.0 F6000		; raise bed back to original height
G90			; absolute mode```

--------------------------------------------------------------

; Y axis homing file

G91			; relative mode
G1 Z4.0 F6000		; lower bed to avoid dragging nozzle over the bed
G1 Y-240 F2000 S1	; move up to 240mm in the -Y direction, stopping if the homing switch is triggered
G1 Y2 F2000		; move slowly  2mm in the +Y direction
G1 Y-10 F 360 S1	; move slowly 10mm in the -Y direction, stopping at the homing switch

G90			; absolute mode
G1 Y0 F2000		; Move head to 0 (corner of bed)
G91			; relative mode
G1 Z-4.0 F6000		; raise bed back to original height
G90			; absolute mode

--------------------------------------------------------------

; Configuration file for Duet WiFi (firmware version 1.17)
; executed by the firmware on start-up

; General preferences
M111 S0 	; Debugging off

; Network
;*** If you have more than one Duet on your network, they must all have different MAC addresses, so change the last digits
;M540 P0xBE:0xEF:0xDE:0xAD:0xFE:0xED ; MAC Address
M550 PFred's Voron	; Machine name
M551 Preprap		; Password
M552 P0.0.0.0 S1	; Enable network and acquire dynamic address via DHCP

G21 		; Work in millimetres
G90 		; Send absolute coordinates...
M83 		; ...but relative extruder moves
M555 P2		; Set firmware compatibility to look like Marlin

M667 S1 ; Select CoreXY mode
M208 X-4 Y-3 Z0 S1	; Set axis minima to match endstop location.  Corner of bed is X0, Y0.
M208 X234 Y232 Z225 S0	; Set axis maxima to end of safe travel.  Corner of bed is X231, X231

; Endstops
M574 X1 Y1 Z0 S1
	; X1 Y1 sets endstops for low end 
	; Z0 because Z endstop is handled in Zprobe setup
	; S1 for active-high endstops. My switches are wired to be NC - Normally Closed.

		; 		**********FRED: Use G31 to set probe height ******************
G31 Z-3.7 	; Should let the system know probe height is -0.7mm so I won't need height adjustment anywhere else
	
M558 P5 X0 Y0 Z1 H3 I0 F120 T18000
	; P5: Set Z probe type to P5.  P5 uses the In and Gnd pins of the "Z PROBE" input (Duet 0.8.5 and Duet WiFi)
	; Z1: Use probe for homing Z axis in addition to bed leveling.
	; H3: Start probing from 3mm under tip (smaller is faster, larger is safer)
	; I0: Do not invert Z-probe signal (could have omitted this parameter)
	; F120: Probing feed rate.  Slower is more accurate.  Want to run some repeatability tests to see if we can speed it up.
	; T18000: Waste no time getting to next probe point.
	; --------------------
	; Note: Probe type P4 apparently works as well.  
	; P4 uses the "E0 STOP" input instead of the "Z PROBE" input, 
	; which is a bit cleaner since it uses the same 3 pin connector as other endstops.

; Driver Configuration
M569 P0 S0	; X Motor goes backwards
M569 P1 S1	; Y Motor goes forwards
M569 P2 S0	; Z Motor goes backwards
M569 P3 S1	; Extruder Motor 1 goes forwards
M569 P4 S0	; Extruder Motor 0 goes backwards

M350 X16    Y16    Z16    E16:16        I1	; 1/16 Microstepping with interpolation (I1)
M92  X80.0  Y80.0  Z400.0 E596.0:596.0		; Steps per mm
M566 X3000  Y3000  Z100   E2000:2000		; Maximum instantaneous speed changes (mm/min) (jerk) 
M203 X18000 Y18000 Z1000  E8000:8000		; Maximum speeds (mm/min)
M201 X3000  Y3000  Z250   E5000:5000		; Accelerations (mm/s^2)
M906 X1700  Y1700  Z1700  E1700:1700    I10 	; Motor currents (mA) and motor idle factor in per cent
M84 S30 ; Set idle timeout

; Heaters
M143 S260 					; Set maximum heater temperature to 260C
M301 H0 S1.00 P10 I0.1 D200 T0.4 W180 B30	; Use PID on bed heater (may require further tuning)
M305 P0 T10000 B4725 C7.06e-8 R4700		; Set thermistor + ADC parameters for heater 0
M305 P1 T100000 B4138 C0 R4700			; Set thermistor + ADC parameters for heater 1

; Tool/Extruder 0
M563 P0 D0 H1	; Tool 0 uses Drive 0 and Heater 1
G10 P0 X0 Y0 Z0 ; Set Tool 0 axis offsets
G10 P0 R0 S0 	; Set Tool 0's initial active and standby temperatures to 0C

; Tool/Extruder 1
M563 P1 D1 H1	; Tool 1 uses Drive 1 and Heater 1
G10 P1 X0 Y0 Z0 ; Set Tool 1 axis offsets
G10 P1 R0 S0 	; Set Tool 1's initial active and standby temperatures to 0C

T0		; select Tool/Extruder 0 by default

; Fans
M106 P0 T45 H1 			; Fan 0 turns on when hot end is >= 45C
;M106 P0 S0.3 I0 F500 H-1 	; Set fan 0 value, PWM signal inversion and frequency. Thermostatic control is turned off
;M106 P1 S1 I0 F500 H1 T45 	; Set fan 1 value, PWM signal inversion and frequency. Thermostatic control is turned on
;M106 P2 S1 I0 F500 H1 T45 	; Set fan 2 value, PWM signal inversion and frequency. Thermostatic control is turned on```

dc42, 2017-03-25:
Fixed in 1.18beta3.

closed 2017-03-25 by dc42

yonkiman, 2017-03-25:
Verified.