Duet3D/RepRapFirmware #86: Linux compile issue, irregular capitalization

marlovdm, 2017-03-23:
I'm trying to compile the firmware on a linux based laptop.
After a couple of hours of troubleshooting (i'm a newbie) i noticed Reprap.h was sometimes referred to as RepRap.h.
On windows this does not matter, however linux throws a load of erros.

dc42, 2017-03-23:
If you let me know which files have the incorrect capitalization in the #include directive then I'll fix it.

dc42, 2017-03-23:
I've corrected the capitalization of the file on github.

marlovdm, 2017-03-23:
woah, that's fast.

marlovdm, 2017-03-23:
MassStorge.h suffers from the same problem.
"Libraties/FatFs/ff.h" does not exist, "Libraties/Fatfs/ff.h" does
Pins_radds.h is referred to as Pins_RADDS.h
With these last two corrected, i succesfully compiled for duet085 and radds.

marlovdm, 2017-03-24:
closing issue

closed 2017-03-24 by marlovdm

dc42, 2017-03-25:
I've corrected the capitalisation of Pins_RADDS.h. MassStorage.h will be fixed at next commit.