Duet3D/RepRapFirmware #87: Feature request: XYZ Compensation

boelle, 2017-03-23:
Hi,

A few indiegogo campaigns are starting to spit out the MK42 heat bed and i wonder if we can see the same style skew compensation as josef's prusa's machine has?

maybe expand it so it says how much skew there is so user has a chance to correct it in hardware first

all i know is that the 9 points are in this file: https://github.com/prusa3d/Prusa-Firmware/blob/MK2/Firmware/mesh_bed_calibration.cpp#L53

dc42, 2017-03-23:
I'll implement this if there is enough demand, but IMO it is a solution to a non-problem. First, very few 3D printer builders fail to get their X and Y axes at right angles. Second, that solution doesn't do anything to measure and correct XZ or YZ skew. Third, RRF has always provided value means to compensate for skew, although you have to measure it manually. See https://duet3d.com/wiki/G-code#M556:_Axis_compensation.

If you still want to see automatic skew correction implemented, please raise it in this forum thread https://www.duet3d.com/forum/thread.php?id=961.

closed 2017-03-23 by dc42

boelle, 2017-03-23:
oh... then i dont get what it is that they do in this video:

https://www.youtube.com/watch?v=rYrLT5G-a9I

dc42, 2017-03-25:
The video demonstrates automatic XY skew correction using an inductive probe and a special bed, plus standard mesh bed levelling. The only thing it does that RRF doesn't do is to make the XY skew calibration automatic. The Prusa machine doesn't do XZ or YZ skew calibration.

I have created a thread at https://www.duet3d.com/forum/thread.php?pid=11360#p11360 to encourage discussion.

boelle, 2017-03-25:
The bed is not special. its a normal zoned heat bed with 9 probe points that you know where is. 

Well XZ and YZ would be awesome extra's to have, could it be done with the same bed?

will go to the thread now

dc42, 2018-08-07:
Manual skew correction has been supported by RepRapFirmware since 2013. Very few printers need XY skew correction, and IMO doing it automatically using a special bed is a gimmick with no real value. I heard that Joseph Prusa himself said it wasn't needed.

You are of course welcome to fork the firmware and add this feature yourself if you wish.