Duet3D/RepRapFirmware #91: Inconsistency in dual Z axis movements

strips, 2017-03-25:
I am trying to enable dual Z axis with separate Z max/high endstops. I get both Z and U motors to stop when their endstops are triggered but when I try to lower Z and U only Z moves.

The issue happens in line 4 in the macro. G1 Z-4  U-4  F600 only the Z motor moves, U remains still.

config.g
```
M584 X0 Y1 Z2:5 U5 E3:4
M574 X1 Y1 Z2 U2 S1
```

macro
```
G91                     ; relative mode
M584 Z2                 ; split Z and U
G1 Z330 U330 F3000 S1   ; move up to 330mm in the +Z direction, stopping if the homing switch is triggered
G1 Z-4  U-4  F600       ; move slowly 4mm in the -Z direction
G1 Z10  U10  F70  S1    ; move slowly 10mm in the +Z direction, stopping at the homing switch
G1 Z-10 U-10 F3000      ; move back again 10mm in the -Z direction
M584 Z2:5               ; joun U to Z again
G90                     ; back to absolute mode
```

dc42, 2017-03-25:
Try adding parameter S2 to line 4.

strips, 2017-03-26:
S2 did the trick. You know whats going on? Should it be like that?

Updated macro works awesome!

```
G91                     ; relative mode
M584 Z2                 ; split Z and U
G1 Z330 U330 F3000 S1   ; move up to 330mm in the +Z direction, stopping if the homing switch is triggered
G1 Z-4  U-4  F600  S2   ; move slowly 4mm in the -Z direction
G1 Z10  U10  F70   S1   ; move slowly 10mm in the +Z direction, stopping at the homing switch
M584 Z2:5               ; joun U to Z again
G1 Z-10      F3000      ; move back again 10mm in the -Z direction
G90                     ; back to absolute mode
```

closed 2020-10-20 by T3P3