Duet3D/RepRapFirmware #92: Feature request: smart ramps support

marlovdm, 2017-04-01:
Hello,
I'm trying to port rrf to my smart ramps.
I know you do not have time to port boards yourself, so i tried it myself.
After two days of messing around i managed to get it to compile, and now i have some time to test the modified code.
Useing the repetier pins.h i managed to port most of the radds pins, sadly i'm in need of some advice regarding the sd card pins.
I modified the SdPciCSPins to fit the board, but i can't get the firmware to use the sdramps.
I can't find the default miso,mosi and sck pins in the config, the sdramps also does not use a sdcarddetect and writeprotect pin.
I assume this is the cause of my problems.
Are my assumptions correct?
If so, how do i fix this?

The full Pins_RADDS.h can be found at: https://pastebin.com/GApBbPwM
Yours truly, marlo

dc42, 2017-04-02:
Make sure you are using the RADDS configuration of the CoreNG project.

The RADDS configuration supports two SPI SD cards. If you want to reduce that to just one, you should edit file asf/conf_sd_mmc.h in the CoreNG project, changing SD_MMC_SPI_MEM_CNT from 2 to 1. Alternatively, define a pin number to use as the CS for the second SD card.

HTH David

dc42, 2017-04-02:
PS - the mosi/miso/sck pins are not configurable in pins.h because they are the SPI0 hardware pins of the ATSAM3X8E chip. They are listed near the bottom of https://www.arduino.cc/en/Hacking/PinMappingSAM3X.

marlovdm, 2017-04-02:
i just checked, and i wasn't using the radds version of coreng, i will report back when i have time to test.
I'm not near my printer often, i expect to be able to work on it in roughly two weeks.

marlovdm, 2017-04-02:
changed SD_MMC_SPI_CNT to 1 and modified the Pins_RADDS.h to only have one value for the cs pin.
I am currenly unable to test it, if anyone wishes to do so they can use the .bin below.

http://s000.tinyupload.com/?file_id=54539682503674058443

marlovdm, 2017-04-14:
had the opportunity to try it out, recieved the following error:
Cannot initialise SD card 0: Card is unusable

The sd card i'm using is a 16gb class 10 sandisk ultra, formatted fat16 with the kossel example loaded

marlovdm, 2017-04-14:
i did some more testing,
I tried another sd card, same result.
regular commands did seem to work, i sucessfully used m119

marlovdm, 2017-04-15:
it turned out the sdramps was faulty.
I ordered a generic full size sd card breakout board.
Hopefully, it will arive tomorrow.
If not, i'll report back in roughly one week

marlovdm, 2017-04-22:
It turned out it was a hardware issue, the new card reader works fine!
I'm doing some basic testing, and will write a guide soon for anyone who wishes to use this.

dc42, 2017-04-22:
I'm glad you got it working.

marlovdm, 2017-04-24:
i'm having om trouble with the z probe.
The radds uses pin 5, and this has to be a adc compatiable pin.
Why does the probe pin have to be a adc pin? can't we digital pins instead?

dc42, 2017-04-24:
RRF supports both analog and digital probes, see https://duet3d.com/wiki/Connecting_a_Z_probe. When using an analog probe, it reduces the probing speed as the trigger point is approached, for better accuracy. I guess you could support a subset of the probe modes if you don't have any spare analog pins.

marlovdm, 2017-04-24:
i do have some spare analog pins, and tested the probe.
i found that if i set the z probe pin number to the spare analog pin, my endstop pins would no longer work.

marlovdm, 2017-04-24:
i set a endstop pin for E1(E0?), and used p4.
And it all worked, trying autocal asap

marlovdm, 2017-05-28:
here my guide and config
[smart ramps.zip](https://github.com/dc42/RepRapFirmware/files/1034149/smart.ramps.zip)
closing this now

dc42, 2017-05-28:
So you did not need to change anything other than Pins_RADDS.h?

marlovdm, 2017-08-17:
Trying to update my firmware, i came across this again.
I changed Pins_RADDS.h and set SD_MMC_SPI_MEM_CNT to 1 in coreng.

The Pins_RADDS.h i sent you is not perfect.
When i assign a pin to the z probe, the endstop pins stop working.
In the version i sent you, the endstops are on the aux header instead.

I will update you on my progress, and send whatever you need.

dc42, 2017-08-17:
Let me know what you end up with so that I can add it to the common source files.

Did you change anything else apart from Pins_RADDS.h?

marlovdm, 2017-08-17:
i did change SD_MMC_SPI_CNT to 1 in coreng, that's all.

I'm having a lot of trouble getting the endstops to work, the pins i want to use:
Xmax: Digital Pin 2
Ymax: Digital pin 15
Zmax: Digital Pin 19
Zprobe: Digital Pin 18
Out of all the things i tried, i was not able to get all endstops to work.
Usually only one pin works.
I will try using analog pin 3 or 4 on the probe, to see if that will fix it.

marlovdm, 2017-08-17:
how do i even define anolog pin numbers?
is analog pin 3 pin 57 or 81 in the config?

dc42, 2017-08-17:
Look at the table near the start of the RADDS version of the variant.cpp file in the CoreNG project. That defines how each pin can be used.

marlovdm, 2017-08-18:
i took a look at the table in variant.h, but it wasn't useful to me.
I've used http://www.robgray.com/temp/Due-pinout.svg and http://www.reprap.org/mediawiki/images/4/4b/SMART_RAMPS_1-4-1_SCH.jpg trying to make this work.

using "const Pin END_STOP_PINS[DRIVES] = { 2, 15, 19, NoPin, NoPin };"
and "const Pin Z_PROBE_PIN = 58;"
Only the Z endstop and probe work.

dc42, 2017-08-18:
Have you changed ENABLE_PIN and STEP_PINS so that they no longer use pins 2 and 15?

marlovdm, 2017-08-18:
yes,
`constPin ENABLE_PINS[DRIVES] =    { 38, 56, 62, 24, 30 };` 
` const Pin STEP_PINS[DRIVES] =      { 54, 60, 46, 26, 36 };` 

i also uploaded repetier, the hardware is fine.

marlovdm, 2017-08-18:
This is the Pins_RADDS i'm currently using
[Pins_RADDS.zip](https://github.com/dc42/RepRapFirmware/files/1235109/Pins_RADDS.zip)

marlovdm, 2017-08-19:
i've been playing a bit with the endstop settings and a multimeter.
It seems like the entire board is pulled up, no matter the S or P parameter in M574.
I'm using normally closed endstops, which shouldn't be affected by it.

marlovdm, 2017-08-19:
I found a workaround, using pins 19 (zmax), 16 and 17.
16 and 17  are on the aux-4 header.

I will probably have to make a wiring guide for this...

I'm going to configure my printer like i normally would and fix any issues that come up.
I'll notify you if i find anything odd, or if i was succesful

marlovdm, 2017-08-22:
Bad news, the native usb port of my due came loose today.
I'll be stuck until drastically improve my soldering skills or buy a new one.

I did notice that when power was supplied, the code would not run unless the reset switch was pressed. The hotend heater would also be on util it is pressed.
I'm unsure what's causing this.

marlovdm, 2017-09-07:
I've succesfully used it on my delta.
I did not test:
- autolevelling
- z probes
- any lcd
- external thermocouple/pt100 boards

Things i noticed:
- make sure the due's 16u2 is up to date, otherwise you will need to reset before the code runs.
- config-override.conf is often ignored.

files: https://drive.google.com/file/d/0B2d61woXZvdIUGZTVjF6Q3BNclU/view?usp=sharing
what have i edited: pins_RADDS.h and the config mentioned earlier.

Wiring guide:
The sd card has to be connected to the spi header on the due.
Sd card cs pin is pin 53 on the aux3 header.
The enstop pins are the following:
Xmax:zmax pin
Ymax: Digital pin 16, aux4
Zmax: Digital pin 17, aux4

I'm considering this done, since the only seller of this board now has sky high shipping prices, and i accidentally got 24v on the thermistor pins rendering my due useless.
I'm going to get a cartridge based hotend, to make sure i don't fry another poor due.

closed 2020-10-20 by T3P3