Duet3D/RepRapFirmware #93: bed.g ignores Y maxima setting from config.g

croadfeldt, 2017-04-21:
Created a bed.g for my printer today, had a typo of Y1291 for one of the probe points. During bed compensation routine, the printer gleefully attempted to fulfill my request of Y1291 when Y maxima was set via M208 Y305. Details below

bed.g
G30 P1 X45 Y1291 Z-99999

config.g
M208 X326 Y305 Z635

dc42, 2017-05-05:
I'm not sure this is really a bug because basically, we expect users to get their configuration files right. There are situations in which probing outside the normal printable area is desirable, for example when probing opposite the towers of a delta printer.

croadfeldt, 2017-05-05:
Fair enough, k, I will close this then. Thanks Dave!

closed 2017-05-05 by croadfeldt